libretro/Lakka #495: RPi3 bluetooth freeze

Kivutar, 2016-10-02:
```
<Zanx> The Rpi3 native BT driver for 4.1 (Sorry is that core? Kernel? :/) causes a system freeze. This is well documented. Upgrading to 4.6 works but I believe a driver patch would fix the issue as well.
<kivutar> ok let me check the kernel version in the last development builds..
<Zanx> I investigated recompiling it but hit a few problems and decided to be patient instead haha.
<Zanx> I DLed Rpi3 stable a few days ago and it was 4.1.
<kivutar> yes, the stable is quite old now
<kivutar> we're still using 4.1
<Zanx> http://lxr.free-electrons.com/diff/drivers/bluetooth/hci_ath.c?v=4.4;diffval=4.1;diffvar=v is a driver comparison
<Zanx> https://www.google.com.au/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#q=raspberry%20pi%20bluetooth%204.1%20freeze top several links are all relevant for extra diagnostics.
```

Kivutar, 2016-10-17:
Fixed.

closed 2016-10-17 by Kivutar

Gregwar, 2016-12-26:
@Kivutar fixed how?
I experience freezes on my rpi3 using dualshock3 joypad, it is related to this?

Kivutar, 2016-12-26:
By this commit https://github.com/libretro/Lakka/commit/a0dd7305888c05c105b07e1867336cab2fd04cb3

And I don't exeprience freezes with my DS3

Gregwar, 2016-12-26:
I am on the 12/12 release img, does it includes this commit?

Kivutar, 2016-12-26:
Yes

Gregwar, 2016-12-26:
Ok, so I am still experiencing freezes, and it seems ok when I use wired
joypad
(Rpi3, 12/12 img and dualshock 3)

Looks like it happens more often on more greedy emulators like N64

Kivutar, 2016-12-26:
Ok that could explain it. Do you often see the rainbow square on the top right corner?

Gregwar, 2016-12-26:
No, never
(I have a 2.5A psu)

I also tried on two boards and two different sd cards (both class 10)

DeFrenZ, 2017-01-06:
I've experienced the same. Disabling "Threaded video" seems to fix it in my case (RPi3 + DS4 for me)... or at least it didn't happen in a while after changing it