libretro/Lakka #501: Intel NUC NUC6i7KYK Blank screen with installer.

woggs, 2016-10-07:
Using an Intel NUC NUC6i7KYK (the one with the skull on it). Lakka runs fine live from USB but screen goes blank and stays that way forever when selecting install. The SSD installed is an M.2 Sata, raw out of the box. I also tried first initializing and installing an MBR on the SSD from Windows with teh same result.

woggs, 2016-10-07:
I should add, I'm perfectly willing to run debug builds and such but I have no idea how to pull and build myself. Willing to learn... Consider me at your disposal with my NUC for testing...

gouchi, 2016-10-07:
Can you try to give the output of `dmesg` and `lsmod` ? You need to access [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) to type those commands.

Thank you.

woggs, 2016-10-07:
I selected "install", the screen goes blank and the machine does not show up on my network, so not sure how to SSH into it. Should I run it live and try the same thing?

woggs, 2016-10-07:
Ouput from running live attached.
[dmesg.txt](https://github.com/libretro/Lakka/files/517216/dmesg.txt)

gouchi, 2016-10-08:
Can you try with [latest release](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161002103858-r21593-g03241bf.img.gz) ?

woggs, 2016-10-08:
Same result... Blank screen forever.

gouchi, 2016-10-08:
Can you check in your BIOS that Devices > Sata > Chipset SATA Mode is set to AHCI ?

woggs, 2016-10-08:
It is indeed AHCI

gouchi, 2016-10-08:
Can you give the ouput of `lspci -nn` ? Because we have enabled AHCI SATA support for the kernel.

woggs, 2016-10-08:
When running live mode from SSH:
[1;32mLakka[1;32m:[1;34m~ [0m# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Sky Lake Host Bridge/DRAM Registers [8086:1910] (rev 0a)
00:02.0 VGA compatible controller [0300]: Intel Corporation Sky Lake Integrated Graphics [8086:193b] (rev 09)
00:08.0 System peripheral [0880]: Intel Corporation Sky Lake Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller [8086:a12f] (rev 31)
00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-H Thermal subsystem [8086:a131] (rev 31)
00:16.0 Communication controller [0780]: Intel Corporation Sunrise Point-H CSME HECI [8086:a13a] (rev 31)
00:17.0 SATA controller [0106]: Intel Corporation Sunrise Point-H SATA Controller [AHCI mode] [8086:a103] (rev 31)
00:1c.0 PCI bridge [0604]: Intel Corporation Sunrise Point-H PCI Express Root Port [8086:a110] (rev f1)
00:1c.1 PCI bridge [0604]: Intel Corporation Sunrise Point-H PCI Express Root Port [8086:a111] (rev f1)
00:1c.2 PCI bridge [0604]: Intel Corporation Sunrise Point-H PCI Express Root Port [8086:a112] (rev f1)
00:1c.4 PCI bridge [0604]: Intel Corporation Sunrise Point-H PCI Express Root Port [8086:a114] (rev f1)
00:1f.0 ISA bridge [0601]: Intel Corporation Sunrise Point-H LPC Controller [8086:a14e] (rev 31)
00:1f.2 Memory controller [0580]: Intel Corporation Sunrise Point-H PMC [8086:a121] (rev 31)
00:1f.3 Audio device [0403]: Intel Corporation Sunrise Point-H HD Audio [8086:a170] (rev 31)
00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-H SMBus [8086:a123] (rev 31)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-LM [8086:15b7] (rev 31)
02:00.0 SD Host controller [0805]: O2 Micro, Inc. Device [1217:8621] (rev 01)
03:00.0 Network controller [0280]: Intel Corporation Wireless 8260 [8086:24f3] (rev 3a)


Kivutar, 2016-10-09:
Here is the new image you can try. It contains the fix attempt of gouchi http://static.kivutar.me/Lakka-Generic.x86_64-devel-20161009201332-r21606-gab3b269.img.gz

woggs, 2016-10-10:
Will try when I get home from work tonight... Thanks!

woggs, 2016-10-10:
Same result. Live runs. Installer goes blank forever. I did notice the my xbox 360 controller stops blinking and is selected as controller 1. I didn't pay attention one way or other before.

woggs, 2016-10-10:
Oh! I just ran the installer again from the new image and this time it kicked into action! Just went through all the menu items, allowed it to install SSH server. Rebooting... And it's live!!

woggs, 2016-10-10:
Is there any further info you would like from my system?

Kivutar, 2016-10-12:
No thanks :)

closed 2016-10-12 by Kivutar