libretro/Lakka #502: SSH enabled in order to work (otherwise, lakka flower for 5-10 minutes between menu selections)

ShockwaveTheFallen, 2016-10-07:
Hardware: Asus Chromebox CN60
Version of Lakka: 1.3.6 (latest off website with Wi-Fi enabled)
Core(s): Reicast, SNES9x, SNES9x Next, SNES9x 2010, Genesis GX, Mupen64, PCSX ReArMeD, FCEUmm
Description of problem: When you switch between games, even within the same core, if you don't have SSH enabled, the system gets hung up on the Lakka flower for a good 5-10 minutes before the game decides to boot up. However, if you have SSH enabled and the terminal is running, loading is instant and fast (less than 2 seconds). Any core, any game. Either Live USB mode or coming off the SSD. Right now, I'm installed off the SSD inside my unit. Below are the logs from the thread I posted in the forums.

ogin as: root
root@192.168.0.20's password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (unofficial) Version: devel-20161002103858-r21593-g03241bf
Lakka git: 03241bf2ccf3fbf6f71d04489e02ae3cee8565e1
Lakka:~ # systemctl stop retroarch
Lakka:~ # aplay -L
null
Discard all samples (playback) or generate zero samples (capture)
hdmi:CARD=HDMI,DEV=0
HDA Intel HDMI, HDMI 0
HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
HDA Intel HDMI, HDMI 1
HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
HDA Intel HDMI, HDMI 2
HDMI Audio Output
default:CARD=PCH
HDA Intel PCH, ALC283 Analog
Default Audio Device
sysdefault:CARD=PCH
HDA Intel PCH, ALC283 Analog
Default Audio Device
front:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
Front speakers
surround21:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
HDA Intel PCH, ALC283 Analog
7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
Lakka:~ # nano /storage/.config/retroarch/retroarch.cfg
Lakka:~ # retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
[... 789 lines cut ...]
[ 4.135678] usbcore: registered new interface driver btusb
[ 4.142016] ieee80211 phy0: Atheros AR9462 Rev:2 mem=0xffffc90001700000, irq=19
[ 4.200790] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 4.305133] random: nonblocking pool is initialized
[ 4.376288] usb 1-4: new full-speed USB device number 4 using xhci_hcd
[ 4.425700] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 4.546760] usb 1-4: string descriptor 0 read error: -22
[ 4.687616] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[ 4.687620] Bluetooth: BNEP filters: protocol multicast
[ 4.687626] Bluetooth: BNEP socket layer initialized
[ 5.027148] wlan0: authenticate with 0c:47:3d:77:7b:68
[ 5.040410] wlan0: send auth to 0c:47:3d:77:7b:68 (try 1/3)
[ 5.042383] wlan0: authenticated
[ 5.043462] wlan0: associate with 0c:47:3d:77:7b:68 (try 1/3)
[ 5.060659] wlan0: RX AssocResp from 0c:47:3d:77:7b:68 (capab=0x411 status=0 aid=2)
[ 5.060772] wlan0: associated
[ 5.060813] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 5.060878] ath: EEPROM regdomain: 0x8348
[ 5.060879] ath: EEPROM indicates we should expect a country code
[ 5.060881] ath: doing EEPROM country->regdmn map search
[ 5.060882] ath: country maps to regdmn code: 0x3a
[ 5.060884] ath: Country alpha2 being used: US
[ 5.060885] ath: Regpair used: 0x3a
[ 5.060887] ath: regdomain 0x8348 dynamically updated by country IE
Lakka:~ #

gouchi, 2016-10-07:
Thank you for the log.

Try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L157) in [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) with
`video_monitor_index = 1`

Don't forget to stop retroarch before making any edits.

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

And reboot to see if it is working.

ShockwaveTheFallen, 2016-10-12:
Almost forgot to tell you, I tried that too, gouchi. Didn't work. I also tried disabling the shader (="0") in the /storage.config/retroarch/retroarch.cfg file and didn't work either. Still takes 5-10 minutes in the choosing of the game/core in order for it to boot up.

ShockwaveTheFallen, 2016-10-12:
It works as long as you leave SSH open (even if  you change the video monitor index as I did here and before), but as soon as you disable SSH/close the terminal, BOOM! Lakka flower and a  5-10 minute startup time between games and menu selections. 

login as: root
root@192.168.0.20's password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (unofficial) Version: devel-20161002103858-r21593-g03241bf
Lakka git: 03241bf2ccf3fbf6f71d04489e02ae3cee8565e1
Lakka:~ # systemctl stop retroarch
Lakka:~ # nano /storage/.config/retroarch/retroarch.cfg
Lakka:~ # retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 3 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 27 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1680x1050) 1680 x 1050, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (1280x960) 1280 x 960, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (1280x800) 1280 x 800, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (1280x768) 1280 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 21: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 22: (720x480i) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 23: (720x480i) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 24: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 25: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 26: (720x400) 720 x 400, 70 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
[... 857 lines cut ...]
RetroArch [WARN] :: [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
RetroArch [INFO] :: [udev]: Plugged pad: Sony PLAYSTATION(R)3 Controller (1356:616) on port #0.
RetroArch [INFO] :: Autodetect: 125 profiles found
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Sony-PlayStation3-DualShock3-Controller-USB.cfg
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event3) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event3) supports 16 force feedback effects.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_video_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_music_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_image_history.lpl
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
[... cut ...]
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
RetroArch [INFO] :: [KMS]: Missed 40 VBlank(s) (Frame: 16816, DRM frame: 54488).
RetroArch [INFO] :: [KMS]: New FB: 1920x1080 (stride: 7680).
RetroArch [INFO] :: [KMS]: Missed 1 VBlank(s) (Frame: 16822, DRM frame: 54494).
RetroArch [ERROR] :: Download failed: Download failed.

markwkidd, 2016-10-22:
Could you clarify: Does this problem occur if you have SSH enabled, but you are not logged in via SSH?

ShockwaveTheFallen, 2016-10-22:
The problem occurs whether or not I have ssh enabled.  It's happened when I'm not logged into the unit via ssh from another pc. So if I log in to the unit from another pc ssh into lakka,  works just beautifully.  But as soon as that connection is broken,  the flower stays on screen for 10 minutes at a time between menu selections/game selections.

Kivutar, 2016-11-22:
Did you enable netplay or something network related in the settings?

thesuffering, 2016-11-22:
I have a similar issue 
http://libretro.com/forums/showthread.php?t=7112

I can say mine the only setting changes were:
Audio Device (to play over hdmi)
Video Output Device (to display via hdmi instead of laptop screen)
Wifi (connecting to ssid and entering password)

netplay and anything else have not been touched.  I supposed the next step I could try loading a rom without connecting to wifi.  I will say, when I had this working before it was via ethernet (the jack broke thus my change to wifi).

ShockwaveTheFallen, 2016-11-23:
It's strange, it's when I specifically restart retroarch with verbose logging in Putty (Win10) is when it performs beautifully. If I leave it open on the Putty window, and the connection doesn't break, Lakka works just fine. As soon as you log off of Putty, the flower returns. It's why the log in this case is useless: I can't generate the log for you and send it to you without being in verbose mode, when it's working. 

Lakka also does NOT operate when in just normal mode without being logged into by Putty and operating in verbose mode. It immediately goes into the Lakka flower as soon as you try to start a game, and stays that way for 5-10 minutes, but the game sometimes does start. 

A little more information for you matt and Kivutar: I installed Ubuntu and Retroarch on the same machine, and RA works just fine with Ubuntu running as the OS. Ubuntu 16.04.1 running 1.3.6 RA and was even running without Xorg: https://github.com/libretro/RetroArch/wiki/Input-drivers-in-Linux-without-Xorg. The terminal window in Ubuntu was making a log of what was occuring in Retroarch (not as detailed as the verbose one in Putty, but similar). So I don't think it's the DRM/KVS. For the record, I also have another CN-60 that I have configured with LibreELEC (Kodi machine) and have RA installed as an add-on app for that machine and it runs on that one, too: https://lime-technology.com/forum/index.php?topic=43836.0. 

Thanks to hunterk and gouchi at the forums, and matt and Kivutar for helping out. hunterk suggested looking at Mesa. Which I don't know what that is. However, you might. :)

ShockwaveTheFallen, 2016-11-23:
To answer your earlier questions about Network settings on my end, no I have not changed any network settings in the menu. I can play with those if you have any suggestions and might do so as a matter of curiosity.

ShockwaveTheFallen, 2016-12-01:
Got it to work: https://libretro.com/forums/showthread.php?t=7514

The problem lies within Mesa and not RA/DRM/KMS, and hunterk said it is a pain in the @$$ to compile that thing to fix it, and I believe him. Since then, the machine has bugged out on me and crashed and isn't worth the headache. Let's close this one out.

closed 2016-12-01 by ShockwaveTheFallen