libretro/Lakka #503: Blank screen when playing ScummVM games

rlay3, 2016-10-09:
Build date: Sep 27 2016
Git version: 03241bf

The ScummVM scanning works fine. However only a few minutes into the game my screen will go blank. Sometimes it gets stuck forever and I'll have to hard reset, other times after 5 seconds it goes back the to Lakka Menu.

So far I've tried monkey, monkey2 and dig, and this has happened to all of them.

I'm running Lakka on a laptop PC, core i5 with a GT 330M graphics chip.

gouchi, 2016-10-09:
Is it possible to provide some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#introduction-to-lakka-logs) ? Thank you.

rlay3, 2016-10-10:
Now that I've got the logs running, it's not happening anymore (of course).

I'll re-open if it does. Thanks.

closed 2016-10-10 by rlay3