libretro/Lakka #506: Implement rpi gpio and wiringpi to Rpi2/3

xprmntlarcade, 2016-10-11:
As discussed in forum:
http://libretro.com/forums/showthread.php?t=6453&page=2&p=48932#post48932

Kivutar is willing to enable this feature "rpi GPIO" and "Wiringpi" so we can connect joysticks and buttons directly into the GPIO Rpi2/3.

Thanks!

xprmntlarcade, 2016-11-10:
HI! 
when could this feature be implemented?
thanks

Kivutar, 2016-11-10:
OK can you try this image please? http://static.kivutar.me/Lakka-RPi2.arm-devel-20161110230408-r21640-ge218f0f.img.gz

xprmntlarcade, 2016-11-10:
Great!, is it the same pinout as: http://libretro.com/forums/attachment.php?attachmentid=643&d=1475334429    ?

Kivutar, 2016-11-10:
I believe so.

Kivutar, 2016-11-10:
I have not added any daemon listening to the GPIO state to emulate a keyboard or a joypad. I just added wiringPi as well as the gpio command line tool.

xprmntlarcade, 2016-11-10:
it doesnt seem to be working, but im making contact touching a ground + LEFT with a wire while in the main menu

xprmntlarcade, 2016-11-11:
Do I need to add the "daemon listening to the GPIO"? If so, how would I do that?
Does this mean this implementation is incomplete? 
Sorry for my ignorance in this matter, I thought it would be working instantly.

Kivutar, 2016-11-15:
Can you check if the gpio command line tool is working?

closed 2016-12-02 by Kivutar