libretro/Lakka #507: Use Fullscreen Mode missing from Video settings with Show Advanced Settings (ON)

wboyo, 2016-10-12:
I upgraded my Intel Celeron 1037U powered PC using the `lakka-update` method for the first time today.  It also automatically purged my `retroarch.cfg` (I did check before the reboot).

After some initial testing I loaded a simple game using PPSSPP and noticed the sound was breaking up as it was struggling to emulate at the full frame rate.

Thinking it may be an updated core, I then realized I was most likely running in Windowed Fullscreen Mode after the upgrade!

Even with the `Settings > Menu > Show Advanced Settings = ON`, the following options (and maybe more I haven't noticed) seem to be missing from `Settings > Video`:

`Use Fullscreen Mode`
`Windowed Fullscreen Mode`
`Windowed Scale`

My workaround to the issue was to putty/ssh on to my Lakka PC and do the following:

`systemctl stop retroarch`
`nano /storage/.config/retroarch/retroarch.cfg`
_Find and Edit:_
`video_fullscreen = "true"`
`video_windowed_fullscreen = "false"`
_^X to Exit and Save_
`systemctl start retroarch`

While the noted menu items are still missing, I definitely have the expected emulation performance back now using Fullscreen Mode again (1080p via HDMI).

:)

Kivutar, 2016-10-12:
Hello, in Lakka there is only one mode: fullscreen. This is why we don't display those settings :)

They are ignored anyway by the context driver.

Kivutar, 2016-10-12:
However I will double check this PPSSPP thing and if the retroarch.cfg is set to that windowed mode.

Kivutar, 2016-10-12:
OMG you're right!

wboyo, 2016-10-12:
Woo Hoo!

Kivutar, 2016-10-12:
This really affects PSP performance?

wboyo, 2016-10-12:
It's the simple Tetris mini game.  I'll get you the FPS comparison if you like.

Not sure if its of any relevance but I also set the following values:

	Custom Viewport Width = 0
	Custom Viewport Height = 0
	Custom Viewport Width = 1920
	Custom Viewport Height = 1080

wboyo, 2016-10-12:
Interesting!  Reporting 60.1 fps in Fullscreen Mode (NTSC) and 50.1 fps in Windowed Fullscreen Mode (oddly PAL it seems?) within PPSSPP.

I am located in a PAL region, so it appears my Panasonic Plasma TV may be recommending 1080p@50Hz over HDMI?

I have seen similar behavior even when connecting MS Windows or Ubuntu based OSes via HDMI to the TV.  I usually manually have to set to 1080p@60Hz.

Hope that helps!

Kivutar, 2016-10-12:
Thanks. @twinaphex could you take a look at this please?

wboyo, 2016-10-12:
Seems to affect others cores too, but not always the fps.

Not conclusive (the fps update was a little erratic) but even the N64 Wave Race (USA) under Mupen64Plus seem to hover at 50fps at the game start in Windows Fullscreen Mode.

Then take a fast scrolling game like NES Super Mario Bros (World) under Nestopia.  While the fps reports 60.1, the scrolling is choppy/looks terrible under Windowed Fullscreen Mode as maybe its attempting 60fps on the windowed PAL 50Hz resolution?!?!

Set the retroarch.cfg back to native Fullscreen Mode and it looks/scrolls perfect!

Let me know if I can be of any more assistance testing wise.

commit 0d2198be (2016-10-17): Force fullscreen mode. Workaround a bug on PAL. Fixes #507
closed 2016-10-17 by Kivutar