libretro/Lakka #508: Broadcom BCM43228 Wifi firmware issue

Marc--, 2016-10-12:
Hi there,

I'm new to Lakka and have a little problem with my HP T610 thin client. It uses a dual WiFi / BT adapter. BT is OK, but Wifi isn't. 

It seems to need the firmware, but i can't install it manually, when i try to copy it to "/lib/firmware/b43/ucode30_mimo.fw", system says "Read-only file system".

Can you help me please ?

Here are the logs.

```
Lakka:~ # lspci -nn |grep BCM
03:00.0 Network controller [0280]: Broadcom Corporation **BCM43228 802.11a/b/g/n [14e4:4359]**
04:00.0 Ethernet controller [0200]: Broadcom Corporation NetLink BCM57781 Gigabit Ethernet PCIe [14e4:16b1] (rev 10)

Lakka:~ # dmesg |grep b43
[    0.374150] pci 0000:00:14.2: reg 0x10: [mem 0xfeb40000-0xfeb43fff 64bit]
[    8.608425] b43-phy0: Broadcom 43228 WLAN found (core revision 30)
[    8.609326] b43-phy0: Found PHY: Analog 9, Type 4 (N), Revision 16
[    8.609342] b43-phy0: Found Radio: Manuf 0x17F, ID 0x2057, Revision 9, Version 1
[    8.610575] b43 bcma0:1: Direct firmware load for b43/ucode30_mimo.fw failed with error -2
[    8.610615] b43 bcma0:1: Direct firmware load for b43/ucode30_mimo.fw failed with error -2
[    8.610672] b43 bcma0:1: Direct firmware load for b43-open/ucode30_mimo.fw failed with error -2
[    8.610700] b43 bcma0:1: Direct firmware load for b43-open/ucode30_mimo.fw failed with error -2
[    8.610707] b43-phy0 ERROR: Firmware file "b43/ucode30_mimo.fw" not found
[    8.610710] b43-phy0 ERROR: Firmware file "b43-open/ucode30_mimo.fw" not found
[    8.610713] b43-phy0 ERROR: You must go to http://wireless.kernel.org/en/users/Drivers/b43#devicefirmware and download the correct firmware for this driver version. Please carefully read all instructions on this website.
```

gouchi, 2016-10-15:
We should update [b43 firmware](https://github.com/libretro/Lakka/tree/lakka/packages/linux-firmware/wlan-firmware) with [this package](http://www.lwfinger.com/b43-firmware/broadcom-wl-5.100.138.tar.bz2) as we are using [kernel > 3.2](http://linuxwireless.org/en/users/Drivers/b43/#If_you_are_using_the_b43_driver_from_3.2_kernel_or_newer:).

Marc--, 2016-10-17:
Hi Gouchi,
Thanks for your reply, i'll wait for next update. Thanks a lot for the job done by the team.
Did i have to close the case ?

gouchi, 2016-10-17:
No problem, you are welcome. Please keep it open, we will update it as soon as we have the new package released.