libretro/Lakka #509: "Icon Shadows" degrades performance on RPi3

duduke, 2016-10-12:
Hi,

I'm using the latest LAKKA on a Raspberry Pi 3 (1080p).
Enabling Icon Shadows under Menu Settings reduces in-menu framerate from 30fps to 10fps. is it by-design, or a "cheaper" shadowing can be used under the Pi to improve performance?

The effect is subtle, so I can just disable it if it cannot be made any faster.

Thanks.

Kivutar, 2016-10-13:
The current shadows are generated by drawing everything a second time. Our code to draw fonts is not optimized for VideoCore. That exaplains the bug.

I tried another way, using an FBO, without success.

Maybe using a shader there would work, I'm not even sure, because adding a pixel shader on a 1920x1080 screen is maybe heavy for the RPi.

Maybe @maister or @aliaspider can help on this.

duduke, 2016-10-13:
Ok, and until they answer, how can I change RetroArch menu resolution?

Kivutar, 2016-10-13:
The best would be to force a lower resolution directly in the config.txt of your first partition. It should also improve performance on some emulators.