libretro/Lakka #510: Xbox one/Pc Razor Wildcat

Soulnz, 2016-10-16:
Any support for the controller, ive got everything running smothlly just this problem holding me back, no disply msg and no lgihts to controller. any ideas out there. tried config tried everything ive read thro the fourms, controllers works fine was using it on my lappy to play my steam games just doesnt like lakka for sum odd reason, 
thanks again
P.S sorry bout grammer got reading and writing issue and half blind lol

gouchi, 2016-10-16:
Did you try to [bind the controller](http://www.lakka.tv/doc/Input-settings/) ?

Is it possible to provide the output of [lsusb](http://www.lakka.tv/doc/Troubleshooting-Lakka/#input-device-logs) ? You will need to access [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface).

Soulnz, 2016-10-16:
heyy there, in settings area i did the binding but havent treid the command line tho just check that out

Soulnz, 2016-10-17:
Anybody able to help. im inching to play games :) i dont know how to get to the command line interface tho i dont understand it due to reading issue

Soulnz, 2016-10-17:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:0a03 Razer USA, Ltd
Bus 001 Device 003: ID 1532:0202 Razer USA, Ltd
Bus 001 Device 002: ID 1038:1390 SteelSeries ApS
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Kivutar, 2016-10-17:
Maybe this controller is not an exact clone of the XBOX One Controller.

Use a keyboard (or another joypad) to navigate to the **Input Settings**. Then go in **Input User 1 Binds** and do a **Bind All**.

Soulnz, 2016-10-17:
Ive tried that it aint working, so confused what could be happening coz the log shows its their just no respose from controller. weird

gouchi, 2016-10-17:
When you plugged your controller, can you provide the output of 
`dmesg | tail`

Thank you.

Soulnz, 2016-10-17:
This is everything i got from the dmesg command
[    0.000000] Linux version 4.6.3-rt8 (kivutar@gcc20) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT RT Tue Sep 13 22:52:09 CEST 2016
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 1088 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000086843fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000086844000-0x0000000086844fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000086845000-0x000000008686efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008686f000-0x00000000868c4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000868c5000-0x0000000086d33fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000086d34000-0x0000000086d49fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000086d4a000-0x00000000879c6fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000879c7000-0x0000000087f99fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000087f9a000-0x0000000087fccfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000087fcd000-0x0000000087ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000088000000-0x00000000880fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000471ffffff] usable
[    0.000000] Notice: NX (Execute Disable) protection cannot be enabled: non-PAE kernel!
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. To be filled by O.E.M./G1.SNIPER B7-CF, BIOS F1 08/14/2015
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x88000 max_arch_pfn = 0x100000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.000000]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.000000]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.000000]   3 base 008C000000 mask 7FFC000000 uncachable
[    0.000000]   4 base 008A000000 mask 7FFE000000 uncachable
[    0.000000]   5 base 0089000000 mask 7FFF000000 uncachable
[    0.000000]   6 base 0088800000 mask 7FFF800000 uncachable
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] found SMP MP-table at [mem 0x000fcab0-0x000fcabf] mapped at [c00fcab0]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x033fffff]
[    0.000000] Base memory trampoline at [c0098000] 98000 size 16384
[    0.000000] BRK [0x02eae000, 0x02eaefff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F05B0 000024 (v02 ALASKA)
[    0.000000] ACPI: XSDT 0x0000000087BFE098 0000B4 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FACP 0x0000000087C20060 00010C (v05 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: DSDT 0x0000000087BFE1E8 021E74 (v02 ALASKA A M I    01072009 INTL 20120913)
[    0.000000] ACPI: FACS 0x0000000087F99F80 000040
[    0.000000] ACPI: APIC 0x0000000087C20170 000084 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FPDT 0x0000000087C201F8 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FIDT 0x0000000087C20240 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[... 617 lines cut ...]
[    4.121245] snd_hda_codec_realtek hdaudioC0D0:    dig-out=0x11/0x1e
[    4.121246] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[    4.121247] snd_hda_codec_realtek hdaudioC0D0:      Front Mic=0x19
[    4.121248] snd_hda_codec_realtek hdaudioC0D0:      Rear Mic=0x18
[    4.121250] snd_hda_codec_realtek hdaudioC0D0:      Line=0x1a
[    4.153751] input: HDA Digital PCBeep as /devices/pci0000:00/0000:00:1f.3/sound/card0/input13
[    4.166963] input: HDA Intel PCH Front Mic as /devices/pci0000:00/0000:00:1f.3/sound/card0/input14
[... cut ...]
[    4.186671] intel_rapl: Found RAPL domain package
[    4.186674] intel_rapl: Found RAPL domain core
[    4.186675] intel_rapl: Found RAPL domain uncore
[    4.186677] intel_rapl: Found RAPL domain dram
[    4.212511] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    4.275441] fuse init (API version 7.24)
[    7.067980] e1000e: eth0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: Rx/Tx
[    7.067984] e1000e 0000:00:1f.6 eth0: 10/100 speed: disabling TSO
[    7.068034] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

Soulnz, 2016-10-18:
Just an extra to add to it, i remembered i had a half working xbox 360 razor sabertooth in a box, i grabbed it out to plug it in and see what happens and it picks up striaght away, cant use it coz sum buttons don't work but shows that the xbox 360 version of the razor controllers work just the xbox wild cat having issues

gouchi, 2016-10-18:
Try to get [lakka system logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/#introduction-to-lakka-logs).

Kivutar, 2016-10-18:
Can you `cat /dev/input/js0` and see if it is writing something when you press the buttons?

Soulnz, 2016-10-21:
i had trouble accessing the command but results show their aint in life when pressing buttons at all :(

Kivutar, 2016-10-21:
Maybe we're lacking the kernel module for this joyad?

Soulnz, 2016-10-21:
Is there anyway round it. i just used sum savings to get new hardware for a custom build for lakka coz im in love with it lol i love my controller hopefully i don't need to get a different one. new zealand is expsive for controllers

Kivutar, 2016-10-21:
Please do a lsusb with just your joypad plugged. I want to identify which line is your joypad to do more research.

Soulnz, 2016-10-22:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:0a03 Razer USA, Ltd
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Kivutar, 2016-10-22:
I find nothing about it, except on a russian website

gouchi, 2016-10-22:
Also can you provide the output of `dmesg` when you controller has been plugged ?

Thank you.

Soulnz, 2016-10-24:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            4.06
  iManufacturer           3 Linux 4.6.3-rt8 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:14.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             6
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
   Port 5: 0000.02a0 5Gbps power Rx.Detect
   Port 6: 0000.02a0 5Gbps power Rx.Detect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
[... 2132 lines cut ...]
f.3/sound/card0/input8
> [    4.125355] input: HDA Intel PCH Rear Mic as /devices/pci0000:00/0000:00:1f
.3/sound/card0/input9
> [    4.127871] input: HDA Intel PCH Line as /devices/pci0000:00/0000:00:1f.3/s
ound/card0/input10
> [    4.130332] input: HDA Intel PCH Line Out Front as /devices/pci0000:00/0000
:00:1f.3/sound/card0/input11
> [    4.131087] input: HDA Intel PCH Line Out Surround as /devices/pci0000:00/0
000:00:1f.3/sound/card0/input12
> [    4.131879] input: HDA Intel PCH Line Out CLFE as /devices/pci0000:00/0000:
00:1f.3/sound/card0/input13
> [    4.132641] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/000
0:00:1f.3/sound/card0/input14
> [    4.133874] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:
00:1f.3/sound/card0/input15
> [    4.134768] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:
00:1f.3/sound/card0/input16
> [    4.135499] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:
00:1f.3/sound/card0/input17
> [    4.268328] FAT-fs (sdb1): Volume was not properly unmounted. Some data may
 be corrupt. Please run fsck.
> [    6.996975] e1000e: eth0 NIC Link is Up 100 Mbps Full Duplex, Flow Control:
 Rx/Tx
> [    6.996979] e1000e 0000:00:1f.6 eth0: 10/100 speed: disabling TSO
> [    6.997024] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

gouchi, 2016-10-28:
We should see something in the dmesg ouput.

Can you try to make a test with another distro, to check if you see something with dmesg ?

Thank you.

Soulnz, 2016-10-29:
ive tried this controller on ubuntu before, all had to install drivers and it worked perfectly, 
so their probaly no way to slove this issue :/

lasers, 2016-10-29:
Try all USB ports.

Soulnz, 2016-10-29:
ill let ya know in mintue if it works

Soulnz, 2016-10-29:
That didnt work either, no input from controller or lights aint coming on, :/ i just finished setting everything up but just need get controller workings, my razor sabertooth works but that has broken buttons so cant use it for gaming :/

lasers, 2016-10-29:
Darn. All USB ports in the back too? Got a USB hub? I have this one controller that won't work if I plug in the front... until I tried again hours later not realizing I plugged in an USB HUB this time... hooked up to the rear, the controller instantly just works.

Soulnz, 2016-10-29:
Haven't tried usb hub yet, I tried all usb front and back and nothing sadly
:/ I'm so despert to get it going all gamed work no controller tho

Kivutar, 2016-12-09:
Maybe you should try one of our LibreELEC based images. It may be something missing in our kernel.