libretro/Lakka #511: Wallpaper lost after reboot and have to quit RA to reload and then it shows. NTFS problem ?

MopheusDG, 2016-10-17:
If you select a PNG file Wallpaper under MENU to use your own background it works, but when you shutdown and restart, the background starts as black. The you have to QUIT/Exit Retroarch and when RA reloads, it appears again.

I think this problem "could" be related to my config files and roms and background and etc being on an NTFS drive, not on default Lakka partitions. Is it possible ? Maybe is something related when the moment the NTFS partitions are mounted not being available when RA starts so no background ? Just my 2 cents..

Thanks.

Kivutar, 2016-10-17:
Did RetroArch crash while launching a game after you set the wallpaper?

When RetroArch crashes, the settings that where changed since the last launch are lost.

MopheusDG, 2016-10-17:
No it didn't. The changes are saved, that's not the problem, cause when I reboot, I don't need to set the wallpaper or create the playlists again, I just need to do a quick QUIT RETROARCH and they just appear. I can create a quick video maybe if it's not clear (sorry my english is not very good).

Kivutar, 2016-10-17:
Yes a video would help thanks. We also speak French.

MopheusDG, 2016-10-17:
I created an small video (by the way this happens not only in my laptop but also on my Brix, both have roms and other stuff on NTFS partition), on the video you can see my laptop booting up, then no wallpaper or playlists, then I just QUIT and everything comes up... weird.

https://youtu.be/VJoD9S9Wlhg

Kivutar, 2016-10-17:
Very weird indeed. Certainly a problem with the hard drive. However, I can't reproduce this bug. Maybe @gouchi can help you.

MopheusDG, 2016-10-17:
I think that to be able to reproduce, playlists and wallpaper png need to be on an NTFS drive, it's the only thing I can think of that could be happening, maybe RA needs the png and playlists available and Lakka has not yet mounted the drives under ROMS ? Anyway... that's all I can help with.

In the meantime, is there anyway to delay RA start or maybe force RetroArch to QUIT after booting the first time so that it can be fixed automatically ?

gouchi, 2016-10-17:
I reproduce the issue but if you mount it with [systemd style](http://wiki.openelec.tv/index.php/Mounting_network_shares#Systemd_NFS_sample_mount), it will work. Just replace kodi.service by retroarch.service

gouchi, 2016-10-17:
I can't reproduce this issue unfortunately. We are talking about NTFS not NFS right ?

MopheusDG, 2016-10-17:
Yes, NTFS for the roms, playlists, etc (I use them also for Windows RA), Lakka Boot and Storage partitions are EXT3/4 not sure, the default ones. Then maybe something else is generating the problem... is there any log I can go over to check ? There should be some error on boot when it cannot load the playlists and wallpaper, right ?

gouchi, 2016-10-18:
Try to get [lakka system logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/#introduction-to-lakka-logs).

MopheusDG, 2016-10-19:
Unfortunatelly I can't do that the way it's required, cause if I boot and enter via SSH when I restart retroarch in verbose mode it works already, so the log will report no problem. I need to be able to launch RA from start in verbose mode, is there any way to do it through the grub menu maybe ?

markwkidd, 2016-10-22:
Hi @MopheusDG could you clarify whether this issue occurs if you start retroarch via SSH but *do not* use verbose mode?

Based on the SSH aspect of this symptom, I'm wondering if it could be related to the issue here: https://github.com/libretro/Lakka/issues/502#issuecomment-255542187

MopheusDG, 2016-10-22:
Unfortunatelly I cannot test this issue anymore, I had a problem and reinstalled so I decided to keep wallpapers and playlists on the EXT4 partitions, what I can confirm is that it's working fine now, with the roms on NTFS drive. What I've been noticing is that after this restart, and after WIFI is enabled (was quite hard on my Gigabyte Brix), sometimes I have some delays between exiting an emulator and going back to the menu, sometimes even 3 or 4 minutes... not sure if it's the wifi (that works really slow not sure why) or something related to something else.... but my main problem dissapeared after forced migration to EXT4. Could try to move those things back and retry reproduce the error if needed....

markwkidd, 2016-10-22:
Thank you for the follow up. It still sounds like there is a bug when you write that there is a delay of 3 or 4 minutes.

I'm a bit curious if that delay would also be affected if you are logged in via SSH or had started retroarch via SSH. That's more than I'm comfortable asking you to do. I'm more of a fan and close reader of this bug tracker than a coder within it!

MopheusDG, 2016-10-23:
The problem is not happening all the time, it seems actually it happens at somepoint, after something goes wrong, and when it starts happening its all the time till you reboot. Let me try to find a little more time for testing it, cause I just reinstalled everything yesterday so I need more testing time to be sure and even get some logs.