libretro/Lakka #516: Audio devices not listed in USB LIVE MODE.

hericstrada, 2016-10-26:
I´m using LAKKA as a LIVE USB from a external HD. When it is plugged in the machine from the first execution of LAKKA, the AUDIO devices are listed if I press the left and right keys, but when I plug to another PC or Notebook, they aren´t listed. If I use Putty and the command aplay -L I can see the names and then write it on the config file, but this way I cant really use the LIVE USB as intended, since the idea is to plug in a friends machinhe and start playing without the need of another device to configure it via Putty.

Since the command aplay - L works very well, I think there are some issue about not updating this list on the Lakka xmb startup.

Thanks for the amazing work!

gouchi, 2016-10-26:
As you are moving to a new PC with different sound card, it will have a different audio device.For now, you should try to put it back to null before you stop Lakka so that next time you can change audio device when you are on a new PC otherwise it will be "stucked" as it can't find the audio device.

gouchi, 2016-11-05:
If you have a keyboard, you can press Enter for Audio Device, it should put it empty.