libretro/Lakka #517: Include rsync

cbluth, 2016-10-27:
Please consider including rsync so that we can rsync content over ssh.

cbluth, 2016-11-12:
I figured out a workaround for this.
Build a statically linked binary: http://www.virtuallyghetto.com/2011/02/how-to-compile-statically-linked-rsync.html

Put it in ~/.local/usr/bin/
Then I did something like this:
`rsync -avPz --rsync-path=/storage/.local/usr/bin/rsync /path/ 192.168.1.12:/storage/roms/`

Enverex, 2016-12-23:
That's not really a solution in the long term. I'd also like to chime in and request that rsync be added as copying larger amounts of content over is a bit of a nightmare otherwise.

cirelli94, 2017-02-13:
Nice workaround @cbluth but I also think Lakka would need rsync!!!! It's a waste of time to use scp for everything...