libretro/Lakka #518: Some Games does not work with Raspberry Pi 2

foxp4v, 2016-11-02:
Hello, some games like R-type, R-Type II, R-Type Leo does not work with the Raspberry Pi version of Finalburn Alpha, when i start them only a black screen is displayed, other games like Wonder Boy simply crash the emulator. All these games works fine with the Windows version. Thanks for all the grat work.

cbluth, 2016-11-03:
Please leave more details about the core/platform these games are being run on, and if you can, the md5sum of each game. I am willing to test them out on my RPi2 setup.

externo6, 2016-11-03:
This is most likely a core issue and not a issue with Lakka.

foxp4v, 2016-11-04:
Probably it's a core issue, if i remind good also an rertopie with the Final Burn Alpha core there are the same problems, instead with the Mame 2003 core works well. I must open an issue otherwhere then?

gouchi, 2016-11-05:
Please try to provide some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka).

Thank you.

Kivutar, 2016-11-08:
Are these games in the list of supported games of FBA?

Kivutar, 2016-11-15:
The games doesn't crash but I get a black screen and RetroArch hangs forever.

foxp4v, 2016-11-26:
With the latest version of Retropie (4.1), the issue is solved, the latest version of lr-fbalpha has fixes for irem hardware on ARM cpus, also Wonderboy works well now

Kivutar, 2016-11-26:
OK I will update lr-fbalpha then.

closed 2016-11-26 by Kivutar