libretro/Lakka #519: No Audio on imx6 CuBox

syepes, 2016-11-05:
Hello,

I have been testing out Lakka on my CuBox and have been fiddling around trying to fix the lack of audio.
Has anyone managed to make the sound work?

Errors
```
# retroarch -v
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [ERROR] :: ALSA: Failed to initialize...
RetroArch [ERROR] :: Failed to initialize audio driver. Will continue without audio.
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
..
..
RetroArch [ERROR] :: Failed to start audio driver. Will continue without audio.
RetroArch [ERROR] :: Failed to start audio driver. Will continue without audio.
RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Resetting undo buffers.
```

Details:
```
# cat /etc/release 
imx6.cuboxi.arm-devel-20161002112747-r21593-g03241bf

# grep "audio_" retroarch.cfg 
audio_driver = "alsa" (Note: I have also tried alsathread)
audio_device = "hw:0,0"
audio_volume = "12.000000"
audio_filter_dir = "/usr/share/audio_filters"
audio_resampler = "sinc"
audio_rate_control_delta = "0.005000"
audio_max_timing_skew = "0.050000"
audio_latency = "64"
audio_block_frames = "0"
audio_out_rate = "44100"
audio_sync = "true"
audio_enable = "true"
audio_mute_enable = "false"
audio_rate_control = "true"
input_audio_mute = "f9"
input_audio_mute_btn = "nul"
input_audio_mute_axis = "nul"
audio_dsp_plugin = ""

# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: imxhdmisoc [imx-hdmi-soc], device 0: i.MX HDMI Audio Tx hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: imxspdif [imx-spdif], device 0: S/PDIF PCM snd-soc-dummy-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=imxhdmisoc
    imx-hdmi-soc, 
    Default Audio Device
sysdefault:CARD=imxhdmisoc
    imx-hdmi-soc, 
    Default Audio Device
hdmi:CARD=imxhdmisoc,DEV=0
    imx-hdmi-soc, 
    HDMI Audio Output
default:CARD=imxspdif
    imx-spdif, 
    Default Audio Device
sysdefault:CARD=imxspdif
    imx-spdif, 
    Default Audio Device
iec958:CARD=imxspdif,DEV=0
    imx-spdif, 
    IEC958 (S/PDIF) Digital Audio Output
```

Thanks for any help.

gouchi, 2016-11-05:
Can you try to put back in the configuration file audio_device="" and [configure it with the gui](http://www.lakka.tv/doc/Audio-settings/). Please don't forget to stop retroarch before editing the configuration file.

Which output you want to use ?

syepes, 2016-11-05:
@gouchi I have just tried that and through the UI the list is empty when trying to set something 
I would like to use the HDMI one , if I am not mistaken it this setting: "hw:0,0" = "hdmi:CARD=imxhdmisoc,DEV=0"

I have just also tried `imx6.cuboxi.arm-devel-20161028093949-r21635-gfb15e2e` and have the same issue.

gouchi, 2016-11-05:
You need to press left or right as specified in [the documentation](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui).

Kivutar, 2016-11-08:
@floreal can you give it a try please? I have no cuboxi to try this. Thank you.

syepes, 2016-11-10:
@gouchi @Kivutar Sorry for the late reply, leaving the audio_device empty and using the left or right buttons worked!

Thanks for the help!

closed 2016-11-10 by syepes