libretro/Lakka #521: Lakka stuck at loading screen when booting

Cr1ms0n, 2016-11-05:
I want to install Lakka on my very old computer.
When I have to launch the live mode it did not work, Lakka stuck at loading screen when booting. (The same problem once install)
I cannot reach cmdline. I suppose that it is an incompatibility with my (very old) graphics card.

Configuration of my computer : 
Intel Pentium 4 2.8Ghz
Ati Radeon 9200 SE 128Mb

It doesn't matter if the problem cannot be solved because my pc is old. (I keep lakka in mind for later)
 
Good luck for the continuation ;)

(I excuse me for the faults but I do not speak English very well)

gouchi, 2016-11-06:
Please is it possible to get some [Lakka system and graphic card logs](http://www.lakka.tv/doc/Troubleshooting-Lakka) ? Maybe your card is [too old for xmb menu](http://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow) ?

You can try to access Lakka command line by [ssh](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#ssh) or [direct access](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/#direct-access).

Thank you.

Cr1ms0n, 2016-11-06:
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 15 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x960) 1280 x 960, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (720x400) 720 x 400, 70 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x1024.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0xb4909588.
RetroArch [INFO] :: [KMS]: New FB: 1280x1024 (stride: 5120).
RetroArch [INFO] :: [GL]: Vendor: Mesa Project, Renderer: Mesa DRI R200 (RV280 5964)  DRI2.
RetroArch [INFO] :: [GL]: Version: 1.3 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x1024
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [ERROR] :: Failed to compile vertex shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Failed to initialize shader, falling back to stock.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [ERROR] :: Failed to compile vertex shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Shader initialization failed.
Segmentation fault

gouchi, 2016-11-06:
Try to [disable the ribbon](http://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow), in the configuration file change to `menu_shader_pipeline = "0"`.

If it still doesn't work, try to change menu to rgui with `menu_driver = "rgui"`.

Don't forget to [stop retroarch before editing the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface).

Cr1ms0n, 2016-11-12:
It still doesn't work (in spite of these two changes)

gouchi, 2016-11-12:
In the configuration file, try to change to `video_monitor_index = "1"`

Cr1ms0n, 2016-11-13:
It still doesn't work (I keep the two previous change)

gouchi, 2016-11-13:
According to this [page](https://www.x.org/wiki/RadeonFeature), your card (R200) doesn't support GLSL.

closed 2016-11-13 by gouchi