libretro/Lakka #522: Random Restarts and Shutdowns

nhosler, 2016-11-08:
I'm not sure if this is a hardware issue or OS issue.  I tend to lean toward OS as I had a Ubuntu live cd on for hours without shutting down or restarting, but when i'm in Lakka, it is doing it quite often. I am not ruling out hardware yet, but wanted to see if it was a known issue with Lakka. My hardware specs are below:

Processor: AMD FX-8350 FX-Series Eight-Core Processor Edition

Motherboard: GIGABYTE GA-990FXA-UD3 AM3+ AMD 990FX

Graphics Card: EVGA GeForce GT 610 1024MB GDDR3

RAM: Cryo-PC 4GB 240-Pin DDR3 SDRAM DDR3 1600

PSU: EVGA 500 Watt 80 Plus Bronze

Hard Drive: 1 TB Western Digital HDD

The processor, motherboard and graphics card were repurposed from an older build, but last I knew, they were working.

Kivutar, 2016-12-09:
This is going to be difficult to debug. @gouchi do you have any idea of what logs we need to ask for there?

gouchi, 2016-12-10:
Try to check the output of `journalctl` ?

Maybe it is related to an option we enabled in kernel config ? What was the Ubuntu version you tried ?