libretro/Lakka #524: Latest EFI builds won't boot

maxengel, 2016-11-10:
With both the x64 and i386 builds from 11/7, a USB stick created with Win32 successfully in Windows 10 only gets to a point where there is a solid cursor in the top-left, and never moves forward.

maxengel, 2016-11-16:
As additional information, creating the USB stick with Rufus and using UEFI and GPT does not help.

Also, other live images work fine, such as the GParted live drive, also created with Rufus.

gouchi, 2016-11-16:
Can you check that the [md5](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161106231917-r21640-ge218f0f-efi.img.gz.md5) or [sha256](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161106231917-r21640-ge218f0f-efi.img.gz.sha256) [checksum](http://www.lakka.tv/doc/Verifying-MD5-Checksums) is correct ?

Thank you.

maxengel, 2016-11-23:
I have confirmed the image is correct. The specific computer is running an Intel Atom processor, if it helps.

gouchi, 2016-11-23:
On which device you try to run Lakka ?

maxengel, 2016-12-25:
It is a [Kangaroo Plus](http://www.kangaroo.cc/kangarooplus/)

gouchi, 2016-12-26:
Can you try to make a test with latest release [64](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161221183631-r21729-g274b40f-efi.img.gz) and [32](http://sources.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20161221183902-r21729-g274b40f-efi.img.gz) bit version? Also, you can try to use [Etcher](https://etcher.io) to create the usb stick.

Thank you.

maxengel, 2016-12-26:
Yes, I'll try it with Etcher. Is there a reason the 64 bit release wouldn't work?

gouchi, 2016-12-29:
No it should work. Did you [disable secure boot](http://forum.kodi.tv/showthread.php?tid=245335&pid=2185490#pid2185490) ?

maxengel, 2016-12-29:
I did disable both secure boot and TPM.