libretro/Lakka #525: Dualshock 4 V2 Not discovered - imx6

syepes, 2016-11-11:
I have just acquired a brand new Dualshock 4 V2 and followed the [procedure](http://www.lakka.tv/doc/Wireless-Dualshock/) but when putting the controller in detection mode “Share + PS Home” I dont see any new bluetooth mac addresses appear.
Note that I had previously borrowed some Dualshock 4 V1 and this worked like a charm.

Has Sony changed more stuff on the V2 that could make it work different?
Anyone know how we could debug this issue?

These are the logs when I connect it to the USB (Not for pairing, just to see if it was recognised):

      -- Logs begin at Sat 2016-12-10 23:24:40 UTC. --
      Nov 11 00:08:33 Lakka kernel: usb 1-1: Product: Wireless Controller
      Nov 11 00:08:33 Lakka kernel: usb 1-1: Manufacturer: Sony Interactive Entertainment
      Nov 11 00:08:33 Lakka kernel: input: Sony Interactive Entertainment Wireless Controller as /devices/soc0/soc.1/2100000.aips-bus/2184000.usb/ci_hdrc.0/usb1/1-1/1-1:1.3/0003:054C:09CC.0001/input/input2
      Nov 11 00:08:33 Lakka kernel: hid-generic 0003:054C:09CC.0001: input,hidraw0: USB HID v1.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-ci_hdrc.0-1/input3
      Nov 11 00:08:33 Lakka kernel: usbcore: registered new interface driver snd-usb-audio
      Nov 11 00:08:33 Lakka Boot[678]: ### Setting up sound card ###
      Nov 11 00:09:14 Lakka bluetoothd[612]: Can't store info for private addressed device /org/bluez/hci0/dev_6E_7D_8C_9E_9B_DC
      Nov 11 00:09:18 Lakka bluetoothd[612]: No cache for 6E:7D:8C:9E:9B:DC
      Nov 11 00:09:44 Lakka bluetoothd[612]: Can't store info for private addressed device /org/bluez/hci0/dev_6E_7D_8C_9E_9B_DC
      Nov 11 00:09:57 Lakka kernel: usb 1-1: USB disconnect, device number 2
      Nov 11 00:11:35 Lakka kernel: usb 1-1: new full-speed USB device number 3 using ci_hdrc
      Nov 11 00:11:35 Lakka kernel: usb 1-1: New USB device found, idVendor=054c, idProduct=09cc
      Nov 11 00:11:35 Lakka kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
      Nov 11 00:11:35 Lakka kernel: usb 1-1: Product: Wireless Controller
      Nov 11 00:11:35 Lakka kernel: usb 1-1: Manufacturer: Sony Interactive Entertainment
      Nov 11 00:11:35 Lakka kernel: input: Sony Interactive Entertainment Wireless Controller as /devices/soc0/soc.1/2100000.aips-bus/2184000.usb/ci_hdrc.0/usb1/1-1/1-1:1.3/0003:054C:09CC.0002/input/input3
      Nov 11 00:11:35 Lakka kernel: hid-generic 0003:054C:09CC.0002: input,hidraw0: USB HID v1.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-ci_hdrc.0-1/input3
      Nov 11 00:11:35 Lakka Boot[807]: ### Setting up sound card ###

Regards and thanks in advance,
Seb

syepes, 2016-11-12:
I fixed the Wired connectivity: 
https://github.com/libretro/retroarch-joypad-autoconfig/pull/257

Kivutar, 2016-11-12:
Merged thanks

syepes, 2016-11-15:
Just for reference the BT mode works correctly on https://github.com/syepes/Lakka-LibreELEC

Kivutar, 2016-12-09:
Any idea of what package is in cause?

syepes, 2016-12-11:
I think it could be the Kernel that is more recent

syepes, 2017-02-20:
I have just re-tested it with the last stable version: Lakka-imx6.cuboxi.arm-devel-20161221185252-r21729-g274b40f.img.gz and its now working!

closed 2017-02-20 by syepes