libretro/Lakka #526: On-Screen Keyboard Cursor is Invisible

cbluth, 2016-11-12:
The addition of the OSK is amazing!.. if I could see the cursor.

I am unsure if I cannot see it because of my dark theme or not, but I have tried a couple themes and I still cannot see it.

I am using Lakka x64 on a NUC-Style computer. I was using the 2nd to the most recent build, did an online update this afternoon, and the new OSK is pretty much unusable after the update.

I have ssh access, let me know if you need logs or other info.

Kivutar, 2016-11-12:
What if you use the monochrome theme?

cbluth, 2016-11-18:
The different icons worked, thanks.
Too bad the other icon theme removes the OSK cursor...

prosthetichead, 2016-11-20:
I havent tested it myself but it should work. The file used to do the border in xmb monchrome is sftp://lakka/tmp/assets/xmb/monochrome/png/key-hover.png it should be possible to copy this file and put it into sftp://lakka/storage/assets/xmb/retroactive/png/key-hover.png for example to get it working for retroactive skin. just change the path you're creating  in assets to whatever the skin is you want to add a key hover too.

cbluth, 2016-12-04:
It looks like there have been improvements with the latest Nov28th update, but some characters on the OSK are still invisible.

Kivutar, 2016-12-04:
They depends on the font. Each theme has its own font, and some of them lacks the characters