libretro/Lakka #529: lakka-update bricks devices

oblitum, 2016-11-13:
First I've run the updater from UI and got #527, still, I thought it would be some compatible image anywy. The updater just reaches 20% and then go back to 0%. I've wait for it to do that 5 times before rebooting. After the 5th time I've rebooted from ssh. On boot it still found a tar file, but md5 failed, so it booted normally.

I've then tried lakka-update from ssh. I didn't display which image it was downloading. On boot the device found a correctly downloaded tar now. It updated the system, and now the system doesn't boot anymore.

oblitum, 2016-11-13:
I'm on a Cubox-i 4x4.

syepes, 2016-11-13:
@oblitum Watchout the lakka-update also has an issue, it does not download the last version.
You should change this on the lakka-update script:
`PROJECT=imx6 -> PROJECT=imx6.cuboxi`

commit 4923fb6d (2016-11-15): Fixes the updater for multi system projects. Fixes #529 and fixes #527
closed 2016-11-15 by Kivutar