libretro/Lakka #530: SSD instability when running Lakka on an Odroid XU4 in a CloudShell enclosure.

raelik, 2016-11-15:
I'm currently running the latest nightly Lakka (Lakka-OdroidXU3.arm-devel-20161106235204-r21640-ge218f0f.tar specifically) on an Odroid XU4 in a CloudShell case (http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143599699669), and I'm experiencing quasi-random instability with the SSD I have connected to the USB->SATA bridge (Genesis GL3321G) provided by the CloudShell (currently a SanDisk X400 256GB model).

On a fresh power cycle, the SSD operates normally, and will continue to do so for a seemingly random amount of time (mintues to hours). Attempting a disk read is what triggers the actual problem, which manifests as a generic "input/output error", with the following in dmesg multiple times over:
```
[ 5108.055177] [c0] sd 0:0:0:0: [sda] Device not ready
[ 5108.058644] [c0] sd 0:0:0:0: [sda]
[ 5108.062212] Result: hostbyte=0x00 driverbyte=0x08
[ 5108.066918] [c0] sd 0:0:0:0: [sda]
[ 5108.070432] Sense Key : 0x2 [current]
[ 5108.074121] [c0] sd 0:0:0:0: [sda]
[ 5108.077686] ASC=0x3a ASCQ=0x0
[ 5108.080628] [c0] sd 0:0:0:0: [sda] CDB:
[ 5108.084575] cdb[0]=0x28: 28 00 1b 01 0a 68 00 00 08 00
[ 5108.089655] [c0] end_request: I/O error, dev sda, sector 453053032
[ 5108.095935] [c2] EXT4-fs warning (device sda1): __ext4_read_dirblock:908: err                                   or reading directory block (ino 14156335, block 0)
[ 5108.112933] [c1] sd 0:0:0:0: [sda] Device not ready
[ 5108.116422] [c1] sd 0:0:0:0: [sda]
[ 5108.116429] Result: hostbyte=0x00 driverbyte=0x08
```

The EXT4-fs warning will vary depending on exactly what action was taken to trigger the error (this was a directory listing, for example).

Signs are pointing to this being an issue resolved in a later kernel version, as this problem does not appear when running the latest Hardkernel-supplied Ubuntu MATE for the XU3/4. It was recommended by the Odroid team to update to their latest 3.10 kernel for the XU3/4, available here: https://github.com/hardkernel/linux/tree/odroidxu3-3.10.y

Kivutar, 2016-11-15:
Ok drop this http://static.kivutar.me/Lakka-OdroidXU3.arm-devel-20161115215459-r21648-g92893a1.tar in your /storage/.update and reboot

raelik, 2016-11-15:
@Kivutar Looks good so far, though my XBox 360 wireless controller isn't being recognized properly now.

raelik, 2016-11-16:
Looking at the other two patches, https://github.com/libretro/Lakka/blob/lakka/projects/OdroidXU3/patches/linux/overlayfs.v18-3.10-rc7.patch probably failed as well, as there were some changes to fs/namei.c that would stop that patch from going in.

raelik, 2016-11-16:
So I spoke too soon, my ROMs stopped loading, and the scan function stopped working.  The same input/output errors popped up in dmesg: https://gist.github.com/raelik/e0f5d7f1e7794bca608e98ae95585752

At a loss now, as I don't get these problems with the HK Ubuntu MATE, which is running the same kernel.

raelik, 2016-11-16:
Actually, the current HK Ubuntu MATE 16.10 release is running 3.10.103.  Not sure how I missed that, thought it was 3.10.104. That said, I don't think that's the issue.

raelik, 2016-11-16:
So, I think I may have uncovered the issue, and it probably isn't a software problem.  This issue only happens when I have the 360 wireless receiver plugged in and am actively using it.  If I use a wired controller it doesn't occur.  That's why I wasn't seeing it on the HK Ubuntu MATE, because I didn't have a controller plugged in while using that.

Kivutar, 2016-11-16:
From all what you said, it looks like a power supply problem.

raelik, 2016-11-16:
If replacing the receiver doesn't fix it, I'll put it on a powered hub and that should resolve it.

raelik, 2016-11-20:
So, I'm at a loss now. I've replaced the wireless receiver AND I have it connected to a powered hub (a 4-port hub with a 2.5A supply, which is more than enough for the receiver, which I found draws 500mA).  I'm still getting these errors when I try to use it.  If I unplug it and use a wired controller, I have no issues.

Kivutar, 2016-11-20:
OK we're going to need the help of the Hardkernel engineers there. Can you try to expose your problem on their forums please?

raelik, 2016-11-20:
I've done that already. Currently, I'm seeing if electrically isolating the 5V pin on the usb cable will help. If so, I'll modify the cable permanently to make it data only.

Kivutar, 2016-11-20:
@mdrjr could you take a look at this please?

raelik, 2016-11-20:
That appears to have done the trick. Apparently the hub I'm using still draws significant power over the bus even when independently powered, but it doesn't NEED to, and it works perfectly fine without that power being available.

raelik, 2016-11-20:
I was banking on this hub not quite obeying the USB spec, but it apparently does. Won't initialize if the 5v connection over the USB cable isn't present.  My only other option at this point is to try putting a resistor on the 5v wire to limit the current to a minimal level.  The hub I'm using appears to have a max power draw of around 400-500mA without the power adapter plugged in.  I should be able to cut that down to 50mA with a 100 ohm 1/4 watt resistor.

diizzyy, 2017-04-01:
This might also be an issue with the GL3321G controller being a bit dodgy, there seems to be random issues with this controller if you do a search on Google. Their older controllers have interesting quirk(s) so it wouldn't surprise me if it also applied to this one. https://irclog.whitequark.org/linux-sunxi/2016-06-27#16843360

raelik, 2017-04-01:
That might be the case, but it only manifests when using that 360 wireless receiver.  I've since stopped using that (switched to bluetooth and PS3 controllers) and have had no issues.