libretro/Lakka #531: Lost Signal after selecting Install

DukeOfNewYork, 2016-11-16:
After burning the ISO to a USB the boot loader will run but just lose video signal when I select Install. However I have been able to get it to boot to a video interface when just selecting live.

I'm trying to install this on a Intel NUC: NUC6i7KYK 
and I'm putting the video out on a HDMI to DVI converter.

I took a quick video of the issue if that helps any.

https://drive.google.com/file/d/0B71bkxzNyEBrZjdVbGRlU3hLRTQ/view?usp=sharing

gouchi, 2016-11-16:
Try to boot in live boot and check which available video output you get then change cmdline (press Tab to edit the entry Run Installer) and append with your information the  [video= parameters](https://wiki.archlinux.org/index.php/Kernel_mode_setting#Forcing_modes_and_EDID).

DukeOfNewYork, 2016-11-17:
So this is weird but it seems to show DP-1 as being used even though it's in HDMI, anyways adding that to video=DP-1 does not seem to change anything and it still will only display video in live mode and not install[image]

gouchi, 2016-11-17:
Can you try to append to the Run Installer cmdline `video=DP-1:e` or `video=DP-1:D` ?

Thank you.

DukeOfNewYork, 2016-11-17:
Both of those still lead to no signal, sorry about that.

gouchi, 2016-11-17:
Can you check in your BIOS that Devices > Sata > Chipset SATA Mode is set to AHCI ?

Your issue should have been [fixed](https://github.com/libretro/Lakka/issues/501).

DukeOfNewYork, 2016-11-17:
My BIOS is set to AHCI, and right now I'm downloading the new imaged linked in that discussion just in case that has a fix that's not in the main branch ( http://static.kivutar.me/Lakka-Generic.x86_64-devel-20161009201332-r21606-gab3b269.img.gz )

DukeOfNewYork, 2016-11-17:
And that image also hasn't helped me, I'm formatting the USB as FAT and using MBR partition scheme for BIOS or UEFI. I feel like I must be missing something obvious.

DukeOfNewYork, 2016-11-24:
So I've got a DP to HDMI cable and now everything is showing up properly, not sure why this thing loves DP so much but I guess this has been solved? If anyone wants any logs to look into this issue just give me a heads up!

closed 2016-11-24 by DukeOfNewYork