libretro/Lakka #532: PC - Installation

schubenn, 2016-11-17:
Hello,

i started working on lakka for three days.
i case of an better performance i installed lakka on a PC.

Installation accomplished without any Problems and starting games for N64 works fine, too. 

But i can't start games for PSX. I've alreadys copied the BIOS for the PSX into the System folder. 
So i can see the games but if i started one the screen only turned black and nothing happend.

another problem is that if i plugged on my USB Joypad (speedlink - SL-6535) nothing happend.
a friend of mine startet a test with my controller on a Cubieboard Machine with Lakka and it works.

i'm very afraid of your comments for solving my problems 

Thanks a lot!

Kivutar, 2016-11-17:
Hello, take a look at our documentation about checksums for the BIOS. It is difficult to find the right BIOS. Some of them are misnamed.
http://www.lakka.tv/doc/PlayStation/

schubenn, 2016-11-17:
Thanks a lot for your post 

UPDATE:

Games Running i played with the Option in INGAME Mode and if i turn on CPU Overclocking it works.

But Lakka didn't work with my controller (speedlink - sl-6535) so any idea here?

Tanks

schubenn, 2016-11-17:
UPDATE:

if i startet the RemoteRetroPad no control LED on the screen turn green if i push a button on the joypad but on the keyboard.

is it possible that i need a driver for the controller?

schubenn, 2016-11-17:
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 005: ID 0e8f:0012 GreenAsia Inc. Joystick/Gamepad
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

The System recognized the Controller

Kivutar, 2016-11-17:
OK for your controller:

Put it in analog mode, and with your keyboard, try to do a manual binding for user #1 in the Input Settings.

We do have drivers for GreenAsia.

schubenn, 2016-11-17:
Thanks for your Posts.

the binding didnt work, no key on the controller is recognized.
Maybe its because the Vendor and Product ID is not listet in the retroarch config files?

Kivutar, 2016-11-17:
try to do this:

cat /dev/input/js0

Then press the keys of your controller, and see if text is appended to the output

schubenn, 2016-11-17:
cat: can't open '/dev/input/js0': No such file or directory

i only have this in /dev/input/ 

Lakka:/dev/input # ls
by-id    by-path  event0   event1   event10  event11  event2   event3   event4   event5   event6   event7   event8   event9   mice

Kivutar, 2016-11-17:
Ok, we're lacking a driver for your pad in the kernel then. I will check that.

Kivutar, 2016-11-17:
Found the bug and fixed it, i'm building a new image for you.

schubenn, 2016-11-17:
thanks a lot

Kivutar, 2016-11-17:
Drop this in your /storage/.update folder http://static.kivutar.me/Lakka-Generic.x86_64-devel-20161117113552-r21652-g04d51b0.tar and reboot

schubenn, 2016-11-17:
Update done

something else to do?

schubenn, 2016-11-17:
running thanks a lot my friend.

Kivutar, 2016-11-17:
The joypad is fixed?

schubenn, 2016-11-17:
yes had to configure it in the input settings but its running fabuolus

Kivutar, 2016-11-17:
OK enjoy ;)

closed 2016-11-17 by Kivutar