libretro/Lakka #535: ScpDriver log of installation opened after every startup

firewater2, 2016-11-19:
Like the title says. Every time I start Windows I get two logs that opens itself automatically. They're identical. The reason why it's two is that I guess I had two installs of DS4Windows at a point so maybe I installed the driver twice or something?

Anyhow this is what appears:

> SCP Driver Installer 1.2.2.175 [05-11-2016 17:42:10]
> 
> Detected - Microsoft Windows 8 Pro (6.2.9200.0 AMD64)
> Selected WIN8 configuration.
> 
> INFO    - ENTER:  DriverPackageInstallW
> ERROR   - Could not open file C:\Users\Cristian\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Startup\System\ScpVBus.inf.
> INFO    - RETURN: DriverPackageInstallW  (0x3)
> 
> Install Succeeded.
> 
> -- Install Summary --
> 
> 
> 
> SCP Driver Installer 1.2.2.175 [05-11-2016 17:44:13]
> 
> Detected - Microsoft Windows 8 Pro (6.2.9200.0 AMD64)
> Selected WIN8 configuration.
> 
> INFO    - ENTER:  DriverPackageInstallW
> ERROR   - Could not open file C:\Users\Cristian\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Startup\System\ScpVBus.inf.
> INFO    - RETURN: DriverPackageInstallW  (0x3)
> 
> Install Succeeded.
> 
> -- Install Summary --

The controller works fine, but this is an annoyance.

gouchi, 2016-11-19:
Hi,
This issue is not related to Lakka, you should post to [ScpToolkit repository](https://github.com/nefarius/ScpToolkit).

Thank you.

closed 2016-11-19 by gouchi

firewater2, 2016-11-19:
Aaaaaah, how did I arrive here!! Thanks...