libretro/Lakka #537: Intel Apollo Lake support

Brunnis, 2016-11-19:
Hi! I just bought the Asrock J4205-ITX, which is an Intel Apollo Lake based mini-ITX board with the Pentium J4205 CPU. The J4205 features the new Goldmont CPU core as well as Intel's Gen 9 graphics, providing a pretty hefty increase in both CPU and GPU power compared to previous Atom derived systems (Bay Trail, Cherry Trail).

The problem I'm having is that Lakka won't start with any of the available images. Here are the combinations I've tried:

**Note:** The tests below are with the r21640 images.
**Note 2:** "CSM" in the table below stands for "Compatibility Support Module", i.e. BIOS emulation mode.
**Note 3:** I've primarily been testing the live mode.

[image]

I've successfully booted Ubuntu 16.10 64-bit in live mode and I've also successfully installed it to disk. I verified that the GPU uses the i915 driver.

Is this a known issue? Maybe an updated Linux kernel is needed? If so, are you able to share your plans for an updated kernel? I'd be willing to try out any test builds if you are able to provide that.

Kivutar, 2016-12-02:
It's a graphic card incompatibility. What graphic card are you using?

Brunnis, 2016-12-02:
I'm using built-in Intel Gen 9 graphics. Graphics acceleration works correctly in Ubuntu 16.10 (running RetroArch in KMS on it right now) and in LibreELEC.

MiniSpoon, 2016-12-11:
Support for that Intel platform was added in mesa 13.0.2. I'm a noob at this, but it looks like that package was merged upstream in OpenELEC a couple weeks ago. Lakka hasn't caught up, yet. I would also like to see mesa updated, for better Vulkan support.

noromamai, 2016-12-14:
I have been trying to get lakka to work (without success) on the same asrock board. i am using the refind bootloader to boot it. so far i have not had much success, but more that is described above. i do not have kernel option "quiet" so see a bit more of the boot process. when booting it takes a long time before it actually starts (dual-booting the same way into libreelec is much faster). when selecting lakka to load, refind will show what options it will pass to the kernel, here it can take up to 10 seconds before the kernel actually starts booting. 9 out of 10 times the boot process will hang for long time at a message about "clock source". sometimes it hangs when detecting the hdd (booting from ssd, hdd is used for storage), but that could be the drive itself (smart test is ok, but it's an old-ish hdd). if this sounds like something other than mesa is causing the problems, i would be happy to test some more and give you more specifics.

noromamai, 2016-12-24:
trying to build lakka with updated mesa. no experience with this, i cloned lakka, changed the mesa version number in the package.mk of graphics/mesa to 13.0.2 and then ty to build for x86_64, but it craps out with:

scons: Reading SConscript files ...
Checking whether the C compiler worksno

	A working C compiler is needed very much.

Makefile:12: recipe for target 'image' failed

gcc is installed (gcc-c++ as well) and also installed dev tools, etc., but it still fails at the same place. not much experince with this, any help/nudge in the right direction would be appreciated.

Kivutar, 2016-12-24:
On what distro are you?

noromamai, 2016-12-24:
I am on Korora 25 (upgraded from 24, Fedora based). The problem was caused by fuse-exfat. Once i disabled all the fuse parts it continued with compiling, but stopped somewhere else. Will keep hacking away and update here how things go.

noromamai, 2016-12-25:
ran to a lot of other stuff that prevented it from compiling. but saw you are also working on Lakka-LibreELEC and that is alerady updated to mesa 13.0.2 so i am tried that. Also not compiling but will move issues to Lakka-LibreELEC project

noromamai, 2017-01-07:
i have managed to compile Lakka-Libretro which has mesa 13.0.2 so should work on this mobo. but did not work for me because refind will not boot the 32-bit efi kernel from 64bit uefi mobo. I had to disable 3 emulators or so to get things to compile. Now trying to compile with a 64bit kernel, but running into other problems. Brunnis, if you are interested in trying out the compiled version, let me know.

Brunnis, 2017-01-10:
Good job! And sure, I can give it a try.

noromamai, 2017-01-11:
two compiles further... you can download the image here: http://shanshan.nl/lakka-libreelec.img.gz the only emulator i removed because compiling problems is nestopia. i have also removed radeon and nvidia drivers. the image will boot up lakka, but when i try to start a game it stops. i did just get it to work by ssh'ing in (root passwd: lakka) and stopping retroarch and restarting in verbose mode (systemctl stop retroarch.service ; retroarch -v). I don't have much time to play around with this, so if you have more luck getting things to work or finding out why it doesnt, please let me know.

Kivutar, 2017-01-11:
I pushed a fix for nestopia a week ago.

noromamai, 2017-01-11:
good to know. i will checkout git again and build with newest sources. i found out that games do run without restarting retroarch from command line, it just takes a very long time (more than a minute) for the emulator to start up. this behaviour is not present after stopping retroarch and starting it again (with or without -v).

noromamai, 2017-01-18:
managed to compile without disabling anything now. if someone needs newer mesa, feel free to download from http://shanshan.nl/lakka/

Brunnis, 2017-01-22:
Sorry, I didn't really get around to testing this. However, I just saw Kivutar's thread on the Libretro forums regarding the updated LibreELEC builds at [le.builds.lakka.tv/Generic.x86_64](http://le.builds.lakka.tv/Generic.x86_64/). I tested on my J4205-ITX with CSM enabled (and selected the non-UEFI item in the boot menu) and it started up as it should! I haven't been able to test any emulators yet but will soon.

Kivutar, 2017-01-22:
OK good to know that the LibreELEC builds are worth something.