libretro/Lakka #539: New example scan script

EthraZa, 2016-11-19:
On https://github.com/libretro/Lakka/wiki/Playlists we have the example bash script. It is a working script but not very smart, so I added a little bit of intelligence to it, but keeped it simple. 
It is basically the same script but now it tests if the rom is not already added to the playlist to not duplicate entries, it does not get lost with spaces in the filenames and now it have some informational output.
So it is still a good example but a little bit more functional for the ones that don't know how to script. 
Hope it help others.

```
#!/bin/bash

#
# Libretro playlist builder
#
# Usage:
# scan.sh "Files path" "Core" "Core name" "Playlist filename" ["Playlists path"]
#
# Example:
#./scan.sh "/storage/roms/arcade/*.zip" "/tmp/cores/fba_libretro.so" "Arcade (FB Alpha - SVN)" "FB Alpha - Arcade Games.lpl" "/storage/playlists/"
#

SAVEIFS=$IFS
IFS=$(echo -en "\n\b")
CA=0
CS=0

echo " Scaning \"$1\" for \"$3\" ROMs and adding to \"$5$4\"... "

for fullpath in $1; do
	touch "$4"
	filename=$(basename "$fullpath") 

	if [ `grep -c "$fullpath" "$4"` -lt 1 ]; then
		CA=$(($CA+1))
		echo "$CA - Adding $filename"
		echo $fullpath >> "$5$4"
		echo "${filename%.*}" >> "$5$4"
		echo $2 >> "$5$4"
		echo $3 >> "$5$4"
		echo "DETECT" >> "$5$4"
		echo $4 >> "$5$4"
	else
		CS=$(($CS+1))
		echo "$CS - Skipping $filename"
	fi
done

echo " Added $CA and skipped $CS \"$3\" ROMs out of $(($CA+$CS)) scanned files to \"$4\" "

IFS=$SAVEIFS
```

cudencuden, 2016-11-19:
you should do a pr at https://github.com/lakkatv/lakka-website

EthraZa, 2016-11-21:
It was my first thought, but I couldn't find how to pr to the wiki pages, so I posted as an issue to let people who can edit the wiki pages to choose to update it.
But if there is a place I can pr to wiki pages, please let me know and I will do it.

Kivutar, 2016-11-21:
Done thanks https://github.com/libretro/Lakka/wiki/Playlists

closed 2016-11-21 by Kivutar