libretro/Lakka #541: 8bitdo controller won't reconnect after a power cycle

maxengel, 2016-11-23:
I can get the 8bitdo controllers working, but they are not automatically reconnected when the power is cycled. Instead, I have to re-pair the controller. Even after connecting and trusting it, it still doesn't reconnect after a power cycle.

duduke, 2016-11-24:
I have the same issue (it might be fixed with the latest 8bitdo firmware), but in any case, I use the following in /storage/.cache/services/bluez.conf:
HIDD_OPTIONS="--connect XX:XX:XX:XX:XX:XX --connect ZZ:ZZ:ZZ:ZZ:ZZ:ZZ"

maxengel, 2016-11-27:
I'm running the latest 8bitdo as of last week. Is there a newer version?

maxengel, 2016-11-27:
also, bluez.conf seems to get overwritten after a reboot...

maxengel, 2016-11-27:
Additionally, it may be related to this: https://libretro.com/forums/showthread.php?t=2347&page=3

'''
Lakka:~ # systemctl status bluetooth● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; disabled; vendor preset: enabled)
   Active: active (running) since Sun 2016-11-27 01:48:30 UTC; 27min ago
 Main PID: 546 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─546 /usr/lib/bluetooth/bluetoothd

Nov 27 01:48:30 Lakka systemd[1]: Started Bluetooth service.
Nov 27 01:48:30 Lakka bluetoothd[546]: Bluetooth daemon 5.37
Nov 27 01:48:30 Lakka bluetoothd[546]: Starting SDP server
Nov 27 01:48:30 Lakka bluetoothd[546]: kernel lacks bnep-protocol support
Nov 27 01:48:30 Lakka bluetoothd[546]: System does not support network plugin
Nov 27 01:48:30 Lakka bluetoothd[546]: Failed to open RFKILL control device
Nov 27 01:48:30 Lakka bluetoothd[546]: Bluetooth management interface 1.9 initialized
Nov 27 01:48:30 Lakka bluetoothd[546]: Failed to obtain handles for "Service Changed" characteristic
'''

duduke, 2016-11-28:
You have enabled the bluetooth service via the lakka settings, right?

maxengel, 2016-11-28:
Yes, I have via the menu

maxengel, 2016-12-04:
this seems to be fixed in the latest nightly, but the wiki should be updated to say entering the PIN may be necessary after the first power cycle of the controller.

Kivutar, 2016-12-04:
I need to try it and understand why we need to pair two times.

lucidlemon, 2016-12-22:
can u install the latest nightly via an update on the pi itself? it kinda always cancels the update for me

maxengel, 2016-12-25:
With the latest nightly it seems that the initial pairing pairs successfully, but the the Bluetooth PIN is only requested the second time I connect.