libretro/Lakka #542: Shelduck

cbluth, 2016-11-27:


Kivutar, 2016-11-27:
Does it sleep when you shutdown lakka? Mine does

prosthetichead, 2016-11-27:
I noticed this also and have been turning my DS3 off when I'm finished playing.
To turn off the DS3 just hold in the PlayStation button for 10sec and the light will go off to turn it back on just press the PlayStation Button again.

It would be really good if it did time out after like 5mins of inactivity though if that is possible?

cbluth, 2016-11-27:
Yes, DS3 *does* power off when lakka is shutdown.

Kivutar, 2016-12-02:
@prosthetichead thanks for the tip.
I have no idea yet how to implement power off on inactivity yet.

prosthetichead, 2016-12-02:
Not sure if it helps But the DS3 does turns itself off if it loses the connection Either through a shutdown/reset or a manual disconnect. After any of these disconnects if you press the PS button in the center it happily and quickly turns back on and connects. So you might not need to actually worry about sending  a turn off signal to the controller?

cbluth, 2016-12-02:
perhaps after 5 minute timeout, or whatever, retroarch detects no activity and sends bluetoothctl disconnect signal. then presumably after bluetooth is disconnected, the controller will shutdown by itself.