libretro/Lakka #543: Updates don't work

Chocobobobo, 2016-11-30:
I am using a RPi3, and update don't work. I have tried both lakka-update using SSH, or dumping the latest update file in the update folder, reboot, but nothing. Current version is still from November 4th, r21640. It should be at least the version from November 28th, r21681.

Kivutar, 2016-11-30:
Ok I try this.

Kivutar, 2016-11-30:
Hello Chocobobobo!

The date in system information is the date of the last build of retroarch, not the date of the last build of lakka.

To know the date of your lakka, do a `lsb_release`.

`lakka-update` will download updates from sources.lakka.tv, while the GUI will download updates from mirror.lakka.tv. sources.lakka.tv contains testing builds, while mirror.lakka.tv contains stable (older) builds.

In both repo, we will look for a file called nightly/.index to know the latest available update. I forgot to update this .index and that explains why you were downloading an old update.

I fixed this, so you can try again, using lakka-update if you want the November 28th

Chocobobobo, 2016-11-30:
Thanks a lot for the quick turnaround.

I could confirm the fix, but also found something "interesting".
I compiled Lakka, based on the latest commit (g895e996).
Using this version, I tried what you mentioned. I could download it and it installed properly.
However, as a final result, I end up with a version older than the one I had installed in the first place.
This is not really an update :)

It might be good when you launch lakka-update to check your current version number against the one in the latest nightly.
Based on the result, you would then have something like "The latest nightly is older than your current version. Are you really sure you want to install it? Y/N" or even "Your version is newer than the latest nightly" or "No update available"

Or changing the name "update" to "get latest nightly" is another way to fix it as well.

Keep it up with the good work :)

cudencuden, 2016-11-30:
+1 chocobobobo

Kivutar, 2016-11-30:
Yes good idea

cudencuden, 2016-11-30:
@Kivutar, is it cool that i try and do this one?

Kivutar, 2016-11-30:
You can try to do this one.

Let me give you some pointers:

To know the date of the current build, you need to either execute `lsb_release`, or open `/etc/os-release`

When we try to access the Lakka Updater, RetroArch will build an URL like this http://mirror.lakka.tv/nightly/RPi2.arm/.index where RPi2.arm is variable.

This file can have more than one line. You can loop over the lines and try to not display a menu entry when the date is older than the current build.

If no lines are displayed, you push an entry like "No updates available" or something like that.

Chocobobobo, 2016-11-30:
Thanks cudencuden and Kivutar, let me know once you have something ready to be tested :)

Kivutar, 2016-11-30:
Ah yes and you need to compile retroarch like this

    make HAVE_LAKKA=1

And create a `/etc/os-release` containing

```
NAME="Lakka"
VERSION="devel-20161127190549-r21681-gb8cc538"
ID="openelec"
VERSION_ID="5.0"
PRETTY_NAME="Lakka (unofficial) - Version: devel-20161127190549-r21681-gb8cc538"
HOME_URL="http://www.openelec.tv"
BUG_REPORT_URL="https://github.com/OpenELEC/OpenELEC.tv"
BUILD_ID="b8cc5380ba94785d48cc2f1af74bd3382571ca15"
OPENELEC_ARCH="RPi2.arm"
OPENELEC_BUILD="unofficial"
```

On your development machine

maemigh, 2017-02-19:
lakka-update downgraded my system. Running from the GUI resulted in the system basically frozen state, hanging on "Downloading .index". There was a weird graphical glitch in that I could only scroll up/down through the update menu, with the text remaining in place, but also showing the text scrolling effect.

When I ran lakka-update from the command line, 

```
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (unofficial) Version: devel-20161221184608-r21729-g274b40f
Lakka git: 274b40f7a5f7a880309a4fe8765e69ac8d8153e1
Lakka:~ # lakka-update
:: Downloading upgrade
Connecting to sources.lakka.tv (51.15.4.11:80)
Lakka-RPi2.arm-devel 100% |*****************************************************************************************************|   206M  0:00:00 ETA
:: Done, you can now reboot
Lakka:~ # shutdown -r now

###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (unofficial) Version: devel-20161128021958-r21681-gb8cc538
Lakka git: b8cc5380ba94785d48cc2f1af74bd3382571ca15
```

You can see that it downgraded from the Christmas release to a build from November. I ran lakka-update again, just in case, but now it kernel panics on boot. The update should probably report that it's on the latest build, or that the current build is newer than what's available rather than continuing and downgrading.

Kivutar, 2017-02-19:
Thanks for the report @maemigh 
I fixed it on the updater. http://mirror.lakka.tv/nightly/RPi2.arm/.index It was due to the recent move to a new server.
Now to fix your setup, do a manual update using the tar of december. The procedure is explained there http://www.lakka.tv/doc/Upgrading-Lakka/

maemigh, 2017-02-19:
hi @kivutar does this also fix the gui freezes in the event that .index is broken? there were some strange graphical glitches happening here. it would be nice if there was some sort of timeout that informed the user that an update couldnt happen. even better if there was a test case for each build that watched for this. :)

Kivutar, 2017-03-05:
This doesn't fix the .index glitch