libretro/Lakka #544: Status Update & Question for Cheating Modul

schubenn, 2016-11-30:
Hello Guys,

after playing lots of stuff with lakka i'm happy to tell everyone that every title i tried work more than good. I'm lovin it. 

Many thanks for this powerfull tool. it combines every old consol in one. Great Work.

After trying and playing many titles in psx n64 gba gb sega an snes, there is one think that i found that still dont work. After the title startet you can enable Cheats, but none of them work. 

I tryied in psx gba und gb but nothing for Xploder Codebreaker or GameGenie work. Still thers something missing in Lakka?

Or is this feature on progress?

Greetings

schubenn, 2016-11-30:
Still intressted in this project? so how can i support you?

Kivutar, 2016-11-30:
Thanks for the kind words :)

The cheats database is not tested, I think they just droped the cheat files in the database repo without checking if they were all working or not.

Maybe you can try to find on the web some cheat files that actually work?

schubenn, 2016-11-30:
is there a chance to get write access in /usr/share/libretro-database/cht/XXXXX/XXXX.cht

if i try to edit the file here putty tell me that this is only a read only file. Logged in as Root

Kivutar, 2016-11-30:
You can override these files in /storage/database