libretro/Lakka #547: 360 controller + Dualshock 3 controller freezes on RPi3

FelipeBR2, 2016-12-02:
I have a X-box 360 controller with the official Microsoft adapter, and a Dual-shock 3 controller. Both controllers work fine, individually, but whenever I open a game with both controllers on, after a while, Lakka just freezes, making me force a shut-down by unplugging the power source.

Also, it seems whenever I plug the Microsoft adapter to my Lakka, it takes over 4 inputs, even though I have only one controller;

Kivutar, 2016-12-02:
You just need more power.
I guess the rainbow square appears at the top right corner just before the freeze?

FelipeBR2, 2016-12-02:
Nope, it doesnt show anything before it freezes. :(

FelipeBR2, 2016-12-02:
It seems to freeze only when I open up a game with multiplayer. (Ex. Street Fighter II, MK III)

Kivutar, 2016-12-28:
I will debug this when I will have the microsoft adapter.

narenniranjan, 2016-12-31:
Hi, I'm having similar hard freezes with the DualShock 3, but only when playing wirelessly. Apparently this was an issue for RetroPie as well: RetroPie/RetroPie-Setup#1449 and it seems to be related to the bluetooth drivers. Apparently its fixed with kernel 4.4 and up, but Lakka is still on 4.1.