libretro/Lakka #549: Works flawlessly, but won't work after rebooting. Error in mount_storage

DanofEarth, 2016-12-04:
First time Lakka user here, I got Lakka 32bit up and running perfectly on an old Windows PC laptop I have. It has 2gb of ram, a Nvidia GTX 7600go and a dual core processor. I have no idea which one.

I have tried re-installing from a clean image and it will always come up perfect the first time, but if I shutdown or reboot I get the following error:

Error in mount_storage: mount_common: Could not mount LABEL=Storage
starting debugging shell... type exit to quit
sh: can't access tty: job control turned off

Google searches didn't help me find anything Lakka specific and I'm not very potent with the command line. I was able to sftp and transfer files over to the laptop just fine prior to having to reboot. 

I also received the following error
mount: mounting /storage on /sysroot/storage failed: Invalid argument

Any help would be greatly appreciated. I love the Lakka I saw briefly. 
[image]

DanofEarth, 2016-12-04:
Also, I just got it working for the first time this evening, it was the very first reboot that caused this issue. Just to clairfy.

DanofEarth, 2016-12-04:
If I type exit I get this screen, I can cycle through the options, but can't actually select anything.
[image]

Kivutar, 2016-12-04:
Hello, are you using the default lakka image untouched? I think we stoped using LABEL=Storage some time ago, that's why I ask.

Can you just try to reinstall it? Looks like a file system corruption.

DanofEarth, 2016-12-04:
Hi there Jean,

Yes, im using the default image from the website. It happened with both 32
and 64 bit versions.

Ive reinstalled it for the 4th time and it seems to be restarting fine,
although, i want to test if it will cause the same error if i remove the
USB. I am writing to hard disc, not launching live, but i think the two may
be connected some how.

The version says it is from sept of 2016.

Thank you,

Dan

Kivutar, 2016-12-08:
And the first time you boot it and it works, it is in installed mode or live mode?

DanofEarth, 2016-12-09:
Installed mode.

Kivutar, 2016-12-09:
I don't get how it can work a first time but not a second time. How many hard drives you have in this PC? Can you try to

    systemctl disable udevil