libretro/Lakka #550: Overlay configuration menu is gone / not available

singularity098, 2016-12-04:
I've installed Lakka on both a Raspberry Pi 2 & 3, and on both systems the Onscreen Overlay menu options are missing. This is using version v1.3.6 of RetroArch in the October 2016 stable Lakka release, and I do see this option in the menus on both Windows and Ubuntu Linux builds of the same version (1.3.6).  This is especially important on a Pi 2 install since it does not have the horsepower for shaders.

Kivutar, 2016-12-06:
Have you tried the crt_pi.glslp?

singularity098, 2016-12-07:
I was not aware of this shader, I'm assuming that it requires very little processing power?  Thanks for mentioning.  Unfortunately the Pi 2 that I was working with is no longer available for me to play with and the owner is not technically inclined so I may need to just keep that in mind for the future.

Kivutar, 2016-12-08:
OK I close this for now then. You can open another issue if this shader is too heavy for a rpi2 but it shouldn't.

closed 2016-12-08 by Kivutar