libretro/Lakka #552: N64 hangs on splash screen

levi-lewis, 2016-12-07:
I'm running lakka on a pc that i just built tonight. When I try to run an N64 rom it hangs on the flower splash screen for a ridiculously long time (20+ minutes) and then finally boots.  Any idea as to why this might be happening?

levi-lewis, 2016-12-08:
If anyone comes looking at this thread, I've started posting in the #487 thread.  Just an update on the issue I have not solved it yet, but have run into an interesting clue as to what's going on.  When it freezes(technically it's not freezing just going really slow) on the flower splash screen if i reboot through SSH and run the rom again it starts up fine.  If I reboot again it freezes and then if i reboot again it runs fine.  It seems like every time i launch a rom after rebooting from when it froze it works fine. I hope this helps someone more knowledgeable than me to find the root of this issue.

Kivutar, 2016-12-08:
I think I got this bug before. It was a texture format issue in RetroArch.