libretro/Lakka #557: No audio on Pi3 through HDMI

ph0rn, 2016-12-08:
Fresh Lakka install (tried latest stable and latest nightly). There's no audio output through HDMI. I've uncommented the "hdmi_drive=2" in the config.txt. That didn't help.
aplay -L only shows the below -- is there supposed to be an HDMI device?

```
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
```

Kivutar, 2016-12-08:
HDMI audio was supposed to work out of the box (especially if you have hdmi_drive=2) so I don't know. Try to switch the audio device in the audio settings using left and right, then restart retroarch between changes.

ph0rn, 2016-12-08:
Yeah I've tried switching between the 'default' and 'sysdefault' in the Audio menu and even rebooted between both - no luck.
I came from recalbox and the audio worked fine after changing the audio to HDMI in EmulationStation menu. I'm not sure what equivalent of that is for retroarch or what it changes exactly.

Kivutar, 2016-12-08:
It's hard to believe that the "hdmi_drive=2" didn't work there. I got audio working on every TV I tried so far.

ph0rn, 2016-12-08:
So to confirm. This is the config.txt on the root of the volume when mounting the SD card right?
Because when I check Lakka, I see there's a:
/flash/config.txt
and
/usr/share/bootloader/config.txt

and only the /flash/confgix.txt has the hdmi_drive change I made. The one in ../bootloader/ still has it commented out. But I'm guessing this is normal though?

Kivutar, 2016-12-08:
The important one is in /flash/config

Try to mount your SD card on your PC, in the first partition, you will see the config.txt

ph0rn, 2016-12-08:
Yep, that's the one I changed and the hdmi_drive=2 is definitely there.
Since the pi3 and Lakka are standard, I'm not sure what else could be going on

Kivutar, 2016-12-09:
what if you use the config.txt of recalbox there?

ph0rn, 2016-12-09:
When using recalbox's config.txt. Lakka will not boot at all.

I did notice that recalbox's config uses hdmi_force_hotplug=1 though. So I tried adding that to Lakka's config.txt. But that did nothing either.

ph0rn, 2016-12-09:
Ok - problem solved. I added "hdmi_force_edid_audio=1" to config.txt

closed 2016-12-09 by ph0rn

Kivutar, 2016-12-09:
OK good to know