libretro/Lakka #564: Adding reicast to XU4 builds

Arnochocobo, 2016-12-13:
Salut Kivutar! J'espere que tout va bien.

Just received my XU4 today, and I love that thing. Not sure why everyone is complaining about 2D tearing because of Mali drivers issues, but Lakka works great on it. 
I have seen this video:
https://www.youtube.com/watch?v=9_LxdO-auP4

Apparently the XU4 has enough horsepower to run Reicast, so I was wondering if you could add it to the XU4 builds?

Thanks!

PS: related topic, is Bsnes completely unplayable on XU4? Even the mercury version?

Kivutar, 2016-12-14:
Hello Arnochocobo,

ça va et toi ?

reicast-libretro is not yet copatible with arm. But it will be in the future. I close this for now :)

closed 2016-12-14 by Kivutar

Arnochocobo, 2016-12-14:
Ca va plutot bien. Leur video est un fake? Depuis le Japon, impossible pour moi de recup la derniere image en un seul morceau et de tester moi-meme, merci Free... Tu peux essayer de jeter un oeil? http://recalbox.remix.free.fr/

Arnochocobo, 2016-12-15:
More and more videos, some people managed to get it working on XU4

https://www.youtube.com/watch?v=VYMisZM18jk

Kivutar, 2016-12-15:
They are not using the libretro version of reicast but the standalone version. In lakka we wait until the libretro team does the port and then we package the core. So we keep things clean with only one configuration file for all the emulators.

Arnochocobo, 2016-12-16:
Understood, merci pour les precisions.