libretro/Lakka #565: Fan control settings on XU4

Arnochocobo, 2016-12-13:
Hello again Kivutar,

I saw that you discussed this topic on the bounty thing with someone and that you received a passive heatsink from HK, but for us not getting presents from them, it would be nice to include this in the XU4 image:

https://www.hiscorebob.lu/2016/10/odroid-xu34-silent-fan-control-in-lakka-openelec/

I checked it, and no more crazyness, this is a way better default solution.

Thanks for looking into this! Everyone's ears and stress level will thank you for sure.

Kivutar, 2016-12-14:
Does it work for both XU3 and XU4?

Arnochocobo, 2016-12-14:
Salut Kivutar,

I can just test on XU4 and according to the site, it has only been tested on XU4. I know there are some fan ID difference between XU3 and XU4 (13 and 14 I believe). I also understand you don't want to create and maintain two different builds for XU3 and XU4. Would be nice if someone with a XU3 could try and let us know the result. But believe me, this is day and night.

cudencuden, 2016-12-14:
xu3 is discontinued.  xu4 could become the official support model

Kivutar, 2016-12-14:
Sure but I own a XU3 so if I grill it then it will become difficult to maintain lakka for XU

Arnochocobo, 2016-12-15:
Worse comes to worse, you can always ask HK for a XU4, I am sure they will have one for you. Having Lakka working on XU4 is the reason why I bought it in the first place, so, mission complete for them with free advertising :)

cudencuden, 2016-12-15:
it is safer to have a xu4 first.

Arnochocobo, 2016-12-15:
"It is dangerous to go alone"

Kivutar, 2016-12-28:
OK I'm working on this now

commit 907551de (2016-12-28): XU fan control. Fixes #565 Fixes #280
closed 2016-12-28 by Kivutar