libretro/Lakka #566: On board Wi-Fi not detected on Nightly Allwinner H3 Orange Pi Lite

manuti, 2016-12-13:
When I go to the menu Wi-Fi I can't see any SSID. But if I connect a Realtek dongle I have Wi-Fi SSID detection but I can't connect to WPA2 network. 
Any ideas? Thanks.

Kivutar, 2016-12-14:
@Ntemis any idea?

manuti, 2016-12-14:
Maybe this thread in the orangepi forum http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=1563 with responses from @jernejsk https://github.com/jernejsk

Ntemis, 2016-12-14:
We need to add realtek 8189FTV driver

jernejsk, 2016-12-14:
rtl8189es driver is already included: https://github.com/lakka-opi/linux/tree/master/drivers/net/wireless/rtl8189es
Fex file seems to be ok. wifi_usbc_* nodes can be removed, because rtl8189es uses SDIO interface, but that is not really the point. Unfortunatelly, I don't have OPiLite at hand, but I will test it on PC Plus (next week or so). If it works there, only few possible explanations are available, because kernel is the same for all boards:
1. fex file misconfiguration
2. HW is broken somehow, e.g. badly attached antenna
3. issue when porting my changes from OE to Lakka

Kivutar, 2016-12-15:
We're not using this tree anymore, we're using https://github.com/libretro/Lakka
So maybe I just forgot to add it

manuti, 2017-01-01:
Continue not working on recent update - Lakka-H3.opilite.arm-devel-20161221194212