libretro/Lakka #567: No core when reboot

Fransebas, 2016-12-14:
I just reinstall lakka (version Lakka-Generic.i386-devel-20161002110423-r21593-g03241bf.img) on a i386 pc and the first time all works fine, but when I reboot It seems horrible, there are black boxes instead of icons and all the titles are en a pixel font, I tried both live and install and is the same on both. In the down-left corner there is a message that says: "1.3.6 no core".

Any idea of what happened?

I've tried to find an older version but can't find one.

Kivutar, 2016-12-14:
I think I know what happened. Somehow, RetroArch doesn't find the hard drive where the assets are.

I suggest you to connect using ssh, do some backup of your files, and do a

    systemctl stop retroarch
    rm -rf .config .cache
    systemctl start retroarch

Fransebas, 2016-12-14:
I finally make it work, turns out that I was connecting the USB in the hd1 instead of hd0, wich is curious because it was booting correctly, it only failed when rebooting, anyway is working perfectly know.

Ps: you rock.

Kivutar, 2016-12-15:
:)

closed 2016-12-15 by Kivutar