libretro/Lakka #569: Wii remotes pair manually but don't connect

rensa, 2016-12-15:
I can use `bluetoothctl` to manually pair Wii remotes to Lakka (on a Raspberry Pi 3) while holding the sync button (using `pair [MAC]`. I can also use `trust [MAC]`. However, the `connect` command fails. Here's the output:

    [bluetooth]# connect 78:A2:A0:AA:3E:9E 
    Attempting to connect to 78:A2:A0:AA:3E:9E
    Failed to connect: org.bluez.Error.Failed
    [bluetooth]# info 78:A2:A0:AA:3E:9E 
    Device 78:A2:A0:AA:3E:9E
	    Name: Nintendo RVL-CNT-01
	    Alias: Nintendo RVL-CNT-01
	    Class: 0x002504
	    Icon: input-gaming
	    Paired: yes
	    Trusted: yes
	    Blocked: no
    	    Connected: no
	    LegacyPairing: yes
	    UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	    UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	    UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	    Modalias: usb:v057Ep0306d8600
    [bluetooth]# info 00:19:1D:8A:C9:36 
    Device 00:19:1D:8A:C9:36
	    Name: Nintendo RVL-CNT-01
	    Alias: Nintendo RVL-CNT-01
	    Class: 0x002504
	    Icon: input-gaming
	    Paired: yes
	    Trusted: yes
	    Blocked: no
	    Connected: no
	    LegacyPairing: yes
	    UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	    UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	    UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	    Modalias: usb:v057Ep0306d3A16

I should note that I'm running `bluetoothctl` over SSH, not by accessing the terminal directly.

Kivutar, 2016-12-15:
I remember that I successfully paired and played with a wiimote 2 month ago. I had to follow a very specific order when entering the commands in bluetoothctl. Another user tested it too, and it fixed the problem for him. I don't remember where I found the right order for the commands.

rensa, 2016-12-16:
Yeah, they seem very fiddly—even getting them to pair required a lot of trial effort w.r.t. timing in `bluetoothctl` :/

Kivutar, 2016-12-28:
I updated the tutorial and checked that the wiimote are working.

closed 2016-12-28 by Kivutar