libretro/Lakka #570: When booting Lakka on my PC, I get stuck at the flower screen.

plutosherry, 2016-12-15:
Lakka:/ # lspci -nnk | grep -A 3 VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation Mobile 945GSE Express Integrated Graphics Controller [8086:27ae] (rev 03)
        Subsystem: Intel Corporation Mobile 945GSE Express Integrated Graphics Controller [8086:27ae]
        Kernel driver in use: i915
00:02.1 Display controller [0380]: Intel Corporation Mobile 945GM/GMS/GME, 943/940GML Express Integrated Graphics Controller [8086:27a6] (rev 03)

when I booting lakka it stuck at the flower screen ? why ?

Kivutar, 2016-12-15:
I think that we can't support this graphic card, if I recal correctly.

@gouchi can you confirm?

plutosherry, 2016-12-16:
But this card (Intel 945GM) can run normally under openelec, but not under the Lakka, I think it should not be a driver problem.
My machine information is: atom N270 / DDR3 1G
It can run smoothly openelec version of Kodi
But can not run Lakka I do not know what the problem
Thx for u response !

Kivutar, 2016-12-16:
openelec uses xorg, while we use drm/kms

gouchi, 2016-12-17:
It should be supported.

Is it possible to get some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

Can you try to append to [the cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) `i915.powersave=0`, it may help to fix your issue.

Thank you.

plutosherry, 2016-12-17:
This is the Lakka logs system
I have been try to append to the cmdine

Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 19 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 4 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x800) 1280 x 800, 58 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x800.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb4907410.
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 8192).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945GME .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x800
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event2 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/event3 as type ID_INPUT_MOUSE.
[... 380 lines cut ...]
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27129, DRM frame: 39764).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27201, DRM frame: 39836).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27273, DRM frame: 39908).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27345, DRM frame: 39980).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27417, DRM frame: 40052).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27489, DRM frame: 40124).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27561, DRM frame: 40196).
RetroArch [INFO] :: [KMS]: Missed 70 VBlank(s) (Frame: 27632, DRM frame: 40267).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27704, DRM frame: 40339).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27776, DRM frame: 40411).
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27848, DRM frame: 40483).

**I have been used ctrl+c to force interrupt loops**

^CRetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27920, DRM frame: 40555).
RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_video_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_music_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_image_history.lpl
RetroArch [INFO] :: [KMS]: Missed 71 VBlank(s) (Frame: 27992, DRM frame: 40627).
RetroArch [INFO] :: Threaded video stats: Frames pushed: 379, Frames dropped: 0.
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.

gouchi, 2016-12-17:
Thank you.

Can you try to edit [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) and set video_monitor_index to "2"

Nano shortcuts (CTRL+o to save and CTRL+x to quit).

plutosherry, 2016-12-17:
First of all thank you very much for your help !
When I set the video_monitor_index=2,  it can work and display menu. 
But if I  reboot the system, i need to manually type "systemctl stop retroarch.service; retroarch -v" under the terminal in cmdline.
It cannot be loaded automatically after each reboot, unless manually enter the command above
why ? And how can I make it run automatically ?

gouchi, 2016-12-17:
It should be written as
`video_monitor_index = "2"`

When you edited the configuration, you stopped RetroArch before editing ?

plutosherry, 2016-12-18:
Yes, I have been modified it.
When lakkas starts, use systemctl stop retroarch.service manually to stop retroarch, then edit the retroarch.cfg with VI and save.
Then use retroarch -v to start retroarch again. It can work & display menu. but if i reboot the system (check the retroarch.cfg found the video_monitor_index= "2" have been set and saved successfully, but the system automatically at startup when loading the retroarch does not take effect), so i need to manually run retroarch service to start & stop operation everytime .why?
thx for U attention!

gouchi, 2016-12-18:
So changing option video_monitor_index doesn't help that much...

plutosherry, 2016-12-18:
How to modify the automatic startup scripts for retroarch services under lakkas? In which position?

plutosherry, 2016-12-18:
After changing the option video_threaded = "false" and reboot the lakkas then the menu is still not work.
When i startup lakkas every time, I should stop retroarch and then run again under the command line.
So I think if I put this "systemctl stop retroarch; retroarch" command written to the script to automatically perform the objective ?

gouchi, 2016-12-18:
Can you make another test with [latest nightly build](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161209100010-r21701-gfa407b3.img.gz) ?

plutosherry, 2016-12-18:
ok I will try to another version
Lakka-Generic.i386-devel-20161209104128-r21701-gfa407b3

plutosherry, 2016-12-18:
when i have been changing the option video_monitor_index="2" & video_threaded="false" then reboot lakkas , **wait for a long times** the menu can display and work.
Why do I need to wait several minutes? Instead of starting as quickly as entering the retroarch command?

plutosherry, 2016-12-18:
when I using "retroarch" to startup retroarch service, the menu canbe quick display 
However, using the "systemctl start retroarch", the menu needs to wait a long time to display
I don't know why?

this is the logfile:
Lakka:~ # retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Dec  8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 19 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 4 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 85 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 85 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (800x600) 800 x 600, 72 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (640x480) 640 x 480, 85 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (640x480) 640 x 480, 73 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (720x400) 720 x 400, 70 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x1024.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0xb4407c78.
RetroArch [INFO] :: [KMS]: New FB: 1280x1024 (stride: 8192).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 945GME .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x1024
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
[... 17 lines cut ...]
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event2 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/event3 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse1 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1280x1024 (stride: 8192).
RetroArch [INFO] :: [KMS]: New FB: 1280x1024 (stride: 8192).

Mwea, 2017-02-17:
Try to edit your retroarch.cfg file 
Disable the video_threaded option and reboot. 
It may fix the problem.

plutosherry, 2017-02-17:
thanks for U response i have been resolved the problem .

发自我的小米手机

closed 2017-02-17 by Kivutar