libretro/Lakka #571: FB Alpha 2012 CPS2 & Dreamcast cores not working anymore

slkzi, 2016-12-15:
I did a clean install of the latest release "Lakka-Generic.x86_64-devel-20161209100010-r21701-gfa407b3" and none of my CPS2 arcade games and Dreamcast start.
All the others (Neo Geo, MAME 2003/2014, NES, SNES, N64 and Mega Drive) works great.

CPS2 game log >>
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Dec 8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x800) 1280 x 800, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x800.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0x7f9ff80c7aa0.
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 5120).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 965GM .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x800
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
[... 531 lines cut ...]
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_TOUCHPAD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse1 as type ID_INPUT_TOUCHPAD.
RetroArch [ERROR] :: [udev] Failed to open device: /dev/input/mouse1 (Inappropriate ioctl for device).
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [ERROR] :: Failed to open libretro core: "/tmp/cores/reicast_libretro.so"
RetroArch [ERROR] :: Error(s): /tmp/cores/reicast_libretro.so: symbol _ZTVNSt7__cxx1119basic_istringstreamIcSt11char_tra itsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_video_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_music_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_image_history.lpl
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 5120).
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 5120).
function is no-op

slkzi, 2016-12-15:
Tested on 2 differents machines with same behaviour.
On another the "Download Cores" new feature is awesome ;-)

Kivutar, 2016-12-15:
Hello, can you try the latest release candidate here please? http://sources.lakka.tv/nightly/
And avoid the "Download Cores", it's not a new feature, it's a deprecated feature that was causing bug.

slkzi, 2016-12-16:
Bonjour Jean-André,
I didn't know about "Download Cores" I though it was new because I never seen it in the past.
Dreamcast problem closed, I simply forgot to transfer BIOSes.

I did an install from scratch with latests release and tried CPS2 cores from 13/12 and 14/12 with same result .
Here is the log  : 
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Dec  8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x800) 1280 x 800, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x800.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0x7f82500c7aa0.
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 5120).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 965GM .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x800
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
[... 92 lines cut ...]
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Dec  8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Loading dynamic libretro core from: "/tmp/cores/fbalpha2012_cps2_libretro.so"
RetroArch [INFO] :: [overrides] no core-specific overrides found at /storage/.config/retroarch/config/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.cfg.
RetroArch [INFO] :: [overrides] no game-specific overrides found at /storage/.config/retroarch/config/FB Alpha 2012 CPS2/cybots.cfg.
RetroArch [INFO] :: Shaders: preset directory: /usr/share/common-shaders/presets
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/cybots.cgp.
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/cybots.glslp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.cgp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.glslp.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/FB Alpha 2012 CPS2/cybots.rmp.
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.rmp.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/cybots.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/cybots.state".
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [libretro ERROR] :: Setting save dir to /storage/savefiles
[FBA] Archive: cybots
retroarch: relocation error: /tmp/cores/fbalpha2012_cps2_libretro.so: symbol _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference

Kivutar, 2016-12-16:
I see that you're using fbalpha2012_cps2_libretro.so
This is not a core packaged in Lakka
If it comes from the core updater, please remove it from your cores directory
I'm going to make sure that this core updater is hidden for real

slkzi, 2016-12-17:
Even after deleteing the fbalpha2012_cps2_libretro.so core coming from libretro.com website it keeps failing.
I'm doing another fresh install and will try the Lakka internal core as 1st attempt.

slkzi, 2016-12-17:
Ok I did a fresh install and there is no available fbalpha2012_cps2_libretro.so provided with Lakka.
I tried to launch a game with FB Alpha and MAME 2003 both failed.
I installed the latest fbalpha2012_cps2_libretro.so available from libretro.com into storage/cores and did a reboot prior to launch any game.
I tried to launch a game using the steps Load Core (fbalpha2012_cps2_libretro.so) then Load Content\Select File and Load Content then Select File and Detect Core. Both ways lead to a failure : 
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Dec  8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Loading default config.
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Incompatible shader for backend gl, clearing...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x800.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7f05e40c7aa0.
RetroArch [INFO] :: [KMS]: New FB: 1280x800 (stride: 5120).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 965GM .
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1280x800
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
[... 45 lines cut ...]
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3
Built: Dec  8 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: fa407b3
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Loading dynamic libretro core from: "/tmp/cores/fbalpha2012_cps2_libretro.so"
RetroArch [INFO] :: [overrides] no core-specific overrides found at /storage/.config/retroarch/config/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.cfg.
RetroArch [INFO] :: [overrides] no game-specific overrides found at /storage/.config/retroarch/config/FB Alpha 2012 CPS2/mmatrix.cfg.
RetroArch [INFO] :: Shaders: preset directory: /usr/share/common-shaders/presets
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/mmatrix.cgp.
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/mmatrix.glslp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.cgp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-shaders/presets/FB Alpha 2012 CPS2/FB Alpha 2012 CPS2.glslp.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/FB Alpha 2012 CPS2/mmatrix.rmp.
[... cut ...]
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/mmatrix.state".
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [libretro ERROR] :: Setting save dir to /storage/savefiles
[FBA] Archive: mmatrix
retroarch: relocation error: /tmp/cores/fbalpha2012_cps2_libretro.so: symbol _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference

slkzi, 2016-12-17:
While looking at the log I can see that even of the core as been put in /storage/cores it keeps trying to launch the core form /tmp/cores. I finally put the core to that path but same result.
Then I put the core in both places and understood that /storage/cores was a redirection point.
Even if I tried previous versions that used to work it keep failing.

cudencuden, 2016-12-17:
well, if you have a core association issue playing from playlist, you have to use text editor to open playlist and remove core association. Also, the game you are trying to launch is Mars Matrix: Hyper Solid Shooting? If yes, I do not think this game is in the latest FB Alpha v0.2.97.39 arcade rom set

Kivutar, 2016-12-17:
Please never use fbalpha2012_cps2_libretro.so in lakka.
The problem may come from the romset. Have you read https://github.com/libretro/Lakka/wiki/Arcade ?

cudencuden, 2016-12-17:
@Kivutar @slkzi , so sorry. mmatrix.zip is found in FB Alpha v0.2.97.39 arcade rom set....the rom does not show up after scanning to create playlist though.

Kivutar, 2016-12-17:
Ok so it's a scanning problem. Our database is still using v0.2.97.38. What about trying the rom from this romset (split).

cudencuden, 2016-12-17:
lol...let me re-download the v0.2.97.38 set.

cudencuden, 2016-12-17:
scanned fba arcade roms v0.2.97.38 split w/o any issue. playlist detected Mars Matrix (000412 USA), Mars Matrix (000412 USA Phoenix Edition) [Bootleg], and Mars Matrix (000412 Japan). Cannot play any of the detected games with FB alpha core...select run then lakka flashes and i return to the playlist...i think this is because my core is v0.2.97.39 core

Kivutar, 2016-12-17:
OK, so she should not have updated FBA to 39 without updating the database to 39.

@larrykoubiak could you help us to update the FBA dat based on v0.2.97.39 please?

cudencuden, 2016-12-17:
well, you would need everything on the same version. core, database, and roms set

TaoPlyPly, 2016-12-18:
@Kivutar Here is the fixed 0.2.97.39 dat & the converted RDB file.

[FB Alpha - Arcade Games.zip](https://github.com/libretro/Lakka/files/659198/FB.Alpha.-.Arcade.Games.zip)

cudencuden, 2016-12-18:
@TaoPlyPly split set? i will test

TaoPlyPly, 2016-12-18:
@cudencuden Yes, based on FBA internal dat generator.

cudencuden, 2016-12-18:
scanned 39 arcade only roms against TaoPlyPly rdb...seems to be picking up the correct amount of games.

hericstrada, 2016-12-19:
Hi guys.  Same problem here,  none of my CPS2 games run after the 09/12 update.  Did I understant right?  I need a new romset?

hericstrada, 2016-12-19:
And I'm talking about Marvel vs Capcom,  X-men Children of Atom,  Darkstalkers etc.  From the. 38 split romset.

cudencuden, 2016-12-19:
if your core is 0.2.97.39 then use 0.2.97.39 roms set. but since rdb is 0.2.97.38..you cannot (easily) create playlist for 0.2.97.39 roms yet...you can still play but have to do Load Content and Detect Core option.

hericstrada, 2016-12-19:
Thanks @cudencuden for the answer! I think the question now is: Did the "Lakka-Generic.x86_64-devel-20161209103132-r21701-gfa407b3.tar" (x64 PC) updated the default FB - Arcade core to the .39? I did not change any default setting, just updated Lakka to this version, and before the update, the .38 set runned normal... @Kivutar, thanks for Lakka!

cudencuden, 2016-12-19:
when you do load core... the ui tells you the core version you are running... lower left hand corner.  Also fba was bumped to .39 in October.  https://github.com/libretro/fbalpha/commit/9f0270491dd25165dc59633c6b3656f18c44f69b

hericstrada, 2016-12-19:
Well,  I would like to revert to the previous core since the. 38 set was working and is enought for me.  @cudencuden,  would you kindly tell me if there is a link where I can download the previous.38 core to replace the new one,   or this is not easy as I thought?

cudencuden, 2016-12-19:
do not know if it is safe. you can try downgrading to previous nighly. http://sources.lakka.tv/nightly/Generic.x86_64/

slkzi, 2016-12-19:
@hericstrada
I think you need to run a clean install of
Lakka-Generic.x86_64-devel-20161028092523-r21635-gfb15e2e

slkzi, 2016-12-19:
@hericstrada
Oops Lakka-Generic.x86_64-devel-20161028092523-r21635-gfb15e2e.tar for .38
FB Alpha core version.

2016-12-19 16:42 GMT+01:00 Jérémie LONKA <jeremie.lonka@gmail.com>:

Rocklima2, 2016-12-24:
I have same problem after update last release of lakka rpi2 , CPS2 game not working more with fba alpha core but using old fba alpha 2012 all CPS2 games working perfect, i waiting fba alpha new release to solve this problem all other games working this core

killer101, 2016-12-28:
Same problem here, no more CPS2 games with the xmas release!

hericstrada, 2016-12-28:
The fba core was updated to .39.  Ive downloaded the .39 set,  regenerated the playlist outside Lakka and everything is working now.

killer101, 2016-12-28:
Yeah, my files won't even work in the .39 windows fba release. I think I'm sure that I'm tested them against the .39 dat file using clrmanepro. Hmmm, maybe I didn't. Okay, getting the .39 romset now.

slkzi, 2016-12-31:
I had to extract .38 core version from 6th November Lakka release (r21640) and push it to latest release (r21729) to be able to run FBA CPS2 games.
I will then need to get .39 compatible romset... A lot of work...