libretro/Lakka #574: Another "stuck at flower screen" issue

hbdfreitas, 2016-12-16:
log file:

-sh: system: not found
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1366x768) 1366 x 768, 60 Hz
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1366x768.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7f39380d58f0.
RetroArch [INFO] :: [KMS]: New FB: 1366x768 (stride: 5632).
Segmentation fault

graphics card log:

00:02.0 VGA compatible controller [0300]: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller [8086:0116] (rev 09)
        Subsystem: Acer Incorporated [ALI] 2nd Generation Core Processor Family Integrated Graphics Controller [1025:0504]
        Kernel driver in use: i915
00:16.0 Communication controller [0780]: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller [8086:1c3a] (rev 04)

If I can provide any more information, please say so!

gouchi, 2016-12-17:
Can you try to [edit the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) and change [the option](http://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow) [menu_shader_pipeline](url) ?

If it still doesn't work try to change option menu_driver to "rgui".

Thank  you.

Mwea, 2017-02-17:
Try to edit your retroarch.cfg file 
Disable the video_threaded option and reboot. 
It may fix the problem.