libretro/Lakka #578: Wii Bluetooth controlled not recognized after reboot

leolite1, 2016-12-18:
I am using the latest nightly build (x64_64 image) image and am trying trying to use Wii controller through a USB Bluetooth adapter.

dmesg initially shows this
```
Lakka:~ # dmesg | grep Blue
[   14.897519] Bluetooth: Core ver 2.21
[   14.897613] Bluetooth: HCI device and connection manager initialized
[   14.897617] Bluetooth: HCI socket layer initialized
[   14.897619] Bluetooth: L2CAP socket layer initialized
[   14.897626] Bluetooth: SCO socket layer initialized
[   15.261489] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.261493] Bluetooth: BNEP filters: protocol multicast
[   15.261497] Bluetooth: BNEP socket layer initialized
```

I paired the controller manually through bluetoothct and trusted the Wii controller as well. Bluetooth status was reported as such
```
Lakka:~ # systemctl status bluetooth
â bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2016-12-16 18:45:28 UTC; 5h 4min ago
 Main PID: 465 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           ââ465 /usr/lib/bluetooth/bluetoothd

Dec 16 18:45:28 Lakka systemd[1]: Started Bluetooth service.
Dec 16 18:45:28 Lakka bluetoothd[465]: Bluetooth daemon 5.37
Dec 16 18:45:28 Lakka bluetoothd[465]: Starting SDP server
Dec 16 18:45:28 Lakka bluetoothd[465]: Failed to open RFKILL control device
Dec 16 18:45:28 Lakka bluetoothd[465]: Bluetooth management interface 1.12 ...ed
Dec 16 18:45:28 Lakka bluetoothd[465]: Failed to obtain handles for "Servic...ic
Hint: Some lines were ellipsized, use -l to show in full.
```

But upon reboot, all that information is lost and I just to do a connect to use the controller
```
Lakka:~ # bluetoothctl
[NEW] Controller 00:0D:88:9E:9E:05 BlueZ 5.37 [default]
[NEW] Device 00:17:AB:39:C6:1B Nintendo RVL-CNT-01
[bluetooth]# connect 00:17:AB:39:C6:1B
Attempting to connect to 00:17:AB:39:C6:1B
[CHG] Device 00:17:AB:39:C6:1B Connected: yes
Connection successful
[Nintendo RVL-CNT-01]#
```

At this point dmesg shows some extra lines

```
[  348.929003] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  348.929011] Bluetooth: HIDP socket layer initialized
[  352.816990] wiimote 0005:057E:0306.0001: unknown main item tag 0x0
[  352.817064] wiimote 0005:057E:0306.0001: hidraw0: BLUETOOTH HID v3a.16 Gamepad [Nintendo RVL-CNT-01] on 00:0d:88:9e:9e:05
[  352.817079] wiimote 0005:057E:0306.0001: New device registered
[  352.868079] wiimote 0005:057E:0306.0001: detected device: Nintendo Wii Remote (Gen 1)
[  352.868688] input: Nintendo Wii Remote Accelerometer as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.3/3-2.3:1.0/bluetooth/hci0/hci0:41/0005:057E:0306.0001/input/input12
[  352.868930] input: Nintendo Wii Remote IR as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.3/3-2.3:1.0/bluetooth/hci0/hci0:41/0005:057E:0306.0001/input/input13
[  352.868998] input: Nintendo Wii Remote as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.3/3-2.3:1.0/bluetooth/hci0/hci0:41/0005:057E:0306.0001/input/input11
```

I have verified that Bluetooth service is enabled and running upon reboot.

Is there a way to force automatic connection (maybe through .config/autostart.sh)?

ervee, 2017-02-27:
I have the same issue. Can pair/connect the Wii Remote. And most of the time it will survive a reboot. But when I shutdown and start Lakka a bit later on, I have to manually enter "connect <MAC> again. Are we doing something wrong or is it a bug?

TrunXGIT, 2017-02-27:
Same happens with 8bitdo controllers.

mmolignano, 2017-04-22:
I'm having the same issue with Wii U Pro controllers. I can connect fine manually, but after restart of my computer, the controllers don't autoconnect. I've paired and trusted the controllers in bluetoothctl, but nothing seems to be working to autoconnect after reboot.

Anyone able to get this to work?

hooby3dfx, 2017-04-29:
I highly suspect this is the same root issue as https://github.com/libretro/Lakka-LibreELEC/issues/83

The timing at which the initial pair/connect/trust may be a factor as mentioned here https://wiki.archlinux.org/index.php/XWiimote#Auto-Reconnect_is_not_working_after_pairing_with_red_sync-button

Could you try using this script to do the pairing and see if persists?
https://github.com/hooby3dfx/lakka-addons/blob/master/btautopair.sh

mmolignano, 2017-05-03:
I'll give it a try and report back.

mmolignano, 2017-05-04:
Alright, thanks to that link on the arch wiki I have gotten it working. No need to use the script either.

The order of commands within bluetoothctl that worked for me are as follows. Make sure that bluetoothctl does not know about the device before you start. If you have already had it connected in the past, enter bluetooth ctl and remove before you start.

```
# bluetoothctl
[bluetooth]# remove <MAC of controller>
```

Now go ahead and pair using the follow sequence.

```
# bluetoothctl
[bluetooth]|# power on
[bluetooth]|# agent on
<press red sync button on controller>
[bluetooth]|# scan on
[bluetooth]|# pair <MAC of controller, use TAB for autocompletion>   # note: we do not explicitly connect yet, we just pair first! (This seems to be the trick to get it to stick) 
[bluetooth]|# connect <MAC of controller>                            # there seems to be a pretty short timeout, so execute this immediately after the pairing command
[bluetooth]|# trust <MAC of controller>
[bluetooth]|# disconnect <MAC of controller>
```

Now pressing a button on the controller to turn it on connects correctly back to the computer. Also when restarting the computer, the controller still auto-connects when pressing a button to turn it on.

ervee, 2017-05-14:
Thank you @mmolignano , this seems to work for me 👍