libretro/Lakka #579: Blank screen for nvidia GXT970 card

paulina-grunwald, 2016-12-19:
Hello,

Lakka boots up to the flower logo but right after the screen goes blank.
I'm using a Nvidia GTX970 video card and a Sony Bravia TV, i've tried mostly the HDMI output but also the DVI with a dvi to hdmi connector to the tv.
I've already tried changing the vlan_monitor_index values 0 to 4 and rebooted after each change, the config file changes persist but the screen still remains blank.
It's the same for live or installation mode.

Logs:
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES
Built: Sep 27 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 03241bf
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch                                                                             .cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retr                                                                             oarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 26 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1920x1080) 1920 x 1080, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (1600x900) 1600 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 13: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 14: (1280x720) 1280 x 720, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 15: (1280x720) 1280 x 720, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 16: (1280x720) 1280 x 720, 30 Hz
RetroArch [INFO] :: [DRM]: Mode 17: (1280x720) 1280 x 720, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 18: (1280x720) 1280 x 720, 24 Hz
RetroArch [INFO] :: [DRM]: Mode 19: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 20: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 21: (720x576) 720 x 576, 50 Hz
RetroArch [INFO] :: [DRM]: Mode 22: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 23: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 24: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 25: (640x480) 640 x 480, 60 Hz
nvc0_screen_create:762 - Error allocating PGRAPH context for M2MF: -16
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory
failed to load driver: kms_swrast
gbm: failed to open any driver (search paths /usr/lib/dri)
gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
failed to load swrast driver
[... 751 lines cut ...]
[  690.277684] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.278748] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.279812] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.280877] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.281934] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.282995] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.284057] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.285118] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.286180] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.287243] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.288305] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.289368] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.290430] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.291494] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.292558] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.293621] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.294683] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.295745] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.296810] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.297874] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.298930] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.299990] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.301051] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
[  690.302122] nouveau 0000:02:00.0: fifo: SCHED_ERROR 08 []
Lakka:~ #

gouchi, 2016-12-20:
Hi,

Thank you for the report. We added support for Maxwell graphic cards but it seems not enough. We will have to update Mesa.

In the time being, is it possible to make a test with [Lakka LE](https://github.com/lakkatv/Lakka-LibreELEC) which is more up to date.

I have made a [test image](http://gouchi.free.fr/Lakka/Lakka-Generic.x86_64-8.0-devel-20161203113326-r24222-g4466c01.img.gz) if you want to try.

Thank you.

loadrunner, 2017-02-21:
Works fine for me on a GTX 960.