libretro/Lakka #581: guidelines for contributing to Lakka repo point to OpenElec material

markwkidd, 2016-12-20:
When creating a new issue in the Lakka github repository, a prominent link instructs the user to "Please review the guidelines for contributing to this repository."

Clicking on this link takes the user to content that seems to be directly copied from the OpenElec source:
https://github.com/libretro/Lakka/blob/lakka/CONTRIBUTING.md

Screenshot:
[image]

gouchi, 2016-12-20:
It seems we should add [ISSUE_TEMPLATE.md](https://github.com/blog/2111-issue-and-pull-request-templates) file in the repository.

markwkidd, 2017-01-17:
PR that addresses the "CONTRIBUTING" document: https://github.com/libretro/Lakka/pull/619

Another PR might be a good idea to add the ISSUE_TEMPLATE.md as mentioned in https://github.com/libretro/Lakka/issues/581#issuecomment-268335510

markwkidd, 2017-01-17:
A second PR to address the ISSUE_TEMPLATE idea: https://github.com/libretro/Lakka/pull/620

closed 2017-02-05 by Kivutar