libretro/Lakka #582: Option to hide/remove the "Remove" item from playlist entries

fiskarz, 2016-12-20:
Entering a ROM within a system's playlist provides three options:  Run, Remove, and Information.  As expected, when "Remove" is selected, the ROM is removed from the playlist, and without confirmation.  Could the option to hide or remove this menu item be added in, so the "Remove" menu option no longer appears?  This pairs well with the other kiosk mode tweaks made to the Oct. build (hide settings menu, etc.)

[image]

shakkalakka, 2016-12-27:
such a hide-feature or confimation would be great!

fiskarz, 2017-05-12:
Just downloaded the new 2.0 and am loving it so far!  The "Remove" option still appears within the playlist entry (ROM) submenu, same as the screenshot from my first post.

Is there a way to remove the "Remove" option that I'm not seeing?  If not, hoping this can get looked at.  I'm sure many people here take time fine tuning their playlists and don't want them to be so easily (or accidentally) tampered with.

fiskarz, 2017-05-12:
Whoops, nevermind!  I found the feature in:

Setting > Playlists > Allow to remove entries

Works as expected - thanks!

closed 2017-05-12 by fiskarz