libretro/Lakka #583: Start rom, lakka berry logo stuck for ~9 minutes.

cbluth, 2016-12-22:
I have a small NUC-style amd64 computer running lakka. It would run lakka perfectly without issue until I moved it to another TV.

Now, on this new TV, it takes about 9-10 minutes to start *any* core, and gets stuck on the lakka berry logo. If I use `systemctl stop retroarch ; retroarch -v`, then it runs perfectly without getting stuck. If reboot or try `systemctl start retroarch`, and then try any rom for any core, then it gets stuck on the berry logo again.

Here is the log when it gets stuck, please note the timestamps:

`Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: CHEEVOS system RAM: (nil) 0
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: CHEEVOS save RAM:   (nil) 0
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: CHEEVOS video RAM:  (nil) 0
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: CHEEVOS RTC:        (nil) 0
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Skipping SRAM load..
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Version of libretro API: 1
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Compiled against API: 1
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Set audio input rate to: 44055.95 Hz.
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Video @ fullscreen
Dec 21 23:51:15 Lakka retroarch[378]: RetroArch [INFO] :: Starting threaded video driver ...
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Found 3 connectors.
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 0 has 14 modes.
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 1 connected: no
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 2 connected: no
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 0: (1360x768) 1360 x 768, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 1: (1920x1080i) 1920 x 1080, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 3: (1280x720) 1280 x 720, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 4: (1280x720) 1280 x 720, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 6: (1440x480i) 1440 x 480, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 60 Hz
Dec 21 23:59:58 Lakka retroarch[378]: RetroArch [INFO] :: [DRM]: Mode 8: (720x480) 720 x 480, 60 Hz
`

budtske, 2017-01-24:
Same problem here, Setting video_monitor_index does not improve this

budtske, 2017-01-28:
At least for me,  [#35](https://github.com/lakkatv/Lakka-LibreELEC/issues/35) is the problem. Removing the network cable makes this issue disappear. I have this issue with Lakka-LE as well as Lakka-OE

cbluth, 2017-01-28:
My computer has several video outputs, and playing with the video settings caused me to get locked out.
Changing `video_monitor_index` via ssh didn't seem to help.