libretro/Lakka #584: lakka not working for oooold pc

turbo511, 2016-12-22:
hey, I just thought that i would ask a question involving my really old hardware. i have recently installed lakka on an old pc that was given to me. this is the most up to date nightly version, the problem is. is that it shows a checklist in the corner and flashes the lakka flower. but nothing else happens.... i was wondering if there is anything that i can do to fix this problem? its an old celeron d based system with integrated intel graphics. if you need any info or can tell me how to get a log file to share. i would be more than glad.

Arnochocobo, 2016-12-22:
You can also buy a pc less than 20 years old, or get a raspberry pi if this is too much, it works great on it and also cost 35 USD. Devs better spend their time on adding new cool features for current hardware than supporting your prehistoric machine.

turbo511, 2016-12-22:
i thought the point of lakka was to repurpose old machines into a retro console, or make it to where you can use cheap hardware "*cough* old" rather than just throw them out. there is no reason for you to be a wiener on a forum when you could have just said "they dont support hardware as old as that, but a good option would be to invest into a raspberry pi, they go as low as 35 USD.

gouchi, 2016-12-22:
Please try to provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka) otherwise it will be difficult to help you.

Thank you.