libretro/Lakka #585: Trouble Configuring Wi-Fi

hansen-m, 2016-12-22:
On a Raspberry Pi 3, installed from NOOBS I'm having a hard time configuring Wi-Fi. With a USB keyboard, as soon as I select the Wi-Fi network and the on-screen keyboard appears I'm not able to actually type the passphrase. Using a game controller I'm able to use the on-screen keyboard to enter it but it I've noticed it is missing several special characters, specifically the apostrophe/single quote.

Kivutar, 2016-12-28:
Yes, we need to imporve this.

There is a bug that prevents us to use keyboard properly in RA in Lakka. It may be linked to a keyboard layout issue. It's difficult to debug.

Now about adding more characters to the OSK, it will be done with time.

luixal, 2017-01-15:
Faced same problem here. Having only one computer (wired to the network) made this a **big** problem.

For anyone getting here, here is what I finally did:

1. Connect the lakka device (i was using a Rpi3) to the network through eth.
2. Get the ip from the Lakka information menu
3.- connect through ssh (from my smartphone) and follow the steps on the [network settings section in the wiki](http://www.lakka.tv/doc/Network-settings/)

Quite annoying, but is working now, only needed to do it once.

As FYI, accessing tty as described [here](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/), didn't work for me. My usb standard keyboard does not work in Lakka using a Rpi3.

Is there any way to set wifi info by editing just a config file? That would have made this really easy. Also, all the problem in my case came from the lack of the "_" and "-" chars in OSK.

zgoda, 2017-03-09:
Because of this there is no way to configure wifi on boards that do not have ethernet and by accident your passphrase contains character like `_` or `-`.

Kivutar, 2017-03-09:
I have pushed a commit in RetroArch to fix this issue. I will update RetroArch in the next update of Lakka and this problem with be fixed.

closed 2017-03-09 by Kivutar

lab313ru, 2017-04-05:
So, when it will be published?

Kivutar, 2017-04-05:
It is published! Check the news on our website :)

lab313ru, 2017-04-05:
But why the link is not the latest at your web site?
http://mirror.lakka.tv/nightly/RPi2.arm/Lakka-RPi2.arm-devel-20161221184608-r21729-g274b40f.img.gz

Kivutar, 2017-04-05:
Yes.